supermerill/SuperSlicer #879: [2.3.55.2] Bridge Flow Causes Major Over-extrusion

CCS86, 2021-01-13:
2.3.55.2 / Win10 x64

Whenever "bridge flow" is triggered on an overhang, even if the "bridge flow ratio" and "above bridge flow ratio" are set to 100%, major over-extrusion is triggered. At first I thought it was a problem with the visualizer, but the issue is confirmed on www.gcodeanalyser.com

Bridge flow off:

[image]

Bridge flow on:

[image]

[image]

[2nd OP Fixt Align.3mf.txt](https://github.com/supermerill/SuperSlicer/files/5811418/2nd.OP.Fixt.Align.3mf.txt)

CCS86, 2021-01-13:
Experimenting with different "bridge flow ratios", I find that a setting of 50% seems to give the correct flow for a 100% width line:

[image]

But SuperSlicer visualizer still shows weirdness:

[image]

robthide37, 2021-01-17:
this has been very documented it has to do with it being impossible to make a line smaller than nozzle width when their is nothing below it to push on and due to the alternate opinions the setting was divorced so you can set it to 0 but still have the fan settings. You will see that depending on the print it is better sometimes but i personnally dont like the bridge flow either but in the case of a true bridge and not a overhang it does benefit the bridging it just is a per case basis cuase if that connected at the top then it would be better to have the bridge flow imo

CCS86, 2021-01-18:
> this has been very documented it has to do with it being impossible to make a line smaller than nozzle width when their is nothing below it to push on and due to the alternate opinions the setting [...]

It's impossible to make a line narrower than nozzle width, even with full support underneath, unless you are filling a gap, bounded on both sides. What does that have to do with this issue?

How could over-extruding a massive amount on an overhang possibly result in a better print? If your printer is tuned slightly towards over-extrusion overhangs are the first thing to suffer. Running with a very slight under-extrusion can potentially improve the result.

But, individual results will vary, so we should talk about definitions. When I see the 'default bridge flow ratio' set to 100%, I assume that this will extrude 100% of the calculated nominal volume, given my defined layer height and line widths. How is 'bridge flow' = 50% extruding the correct amount of filament a desired behavior?

CCS86, 2021-01-18:
The issue isn't just on overhangs though. When bridge flow is triggered on a true bridge, having the bridge flow ratio set to 100% results in massive over-extrusion as well:

[image]

CCS86, 2021-01-18:
Also the 'above the bridges' flow ratio seems to be non-functional. Here it is set to 120%:

[image]

robthide37, 2021-01-18:
i get it trust me but what nozzle size and what layer height cause like i said imo i do not use brisge flow unless i am at layer height => 50% nozzle width otherwise it does what you say but once againthink of it . How would you print a bridge that is not  the width of the nozzle in layer height as well ? If you try to print at .1 layer with a .4 nozzle it would just end up with a failed bridge or very underextruded cause it has nothing to push against to make it full width while only beign .1 height . Instead you would get a round extrusion somewhere between the two

CCS86, 2021-01-18:
In this case, I am printing 0.2 layers, with a 0.4 nozzle.

Of course, bridges will print a somewhat round extrusion, due to the lack of support. It is impossible to get a 1st bridge layer that doesn't look like under-extrusion. The whole point of a bridge is to span a gap, with minimal sag, providing a platform to recover on. It will always be under-extruded, which is why printing the next layer with a slight over-extrusion can help recover faster.

The general consensus for good bridges is to print them with nominal flow rate, or a very slight under-extrusion to provide extra tension. Printing with a bunch of over-extrusion like SuperSlicer is doing here will sag the crap out of them.

supermerill, 2021-01-23:
currently, bridge flow (in mm3/mm) is computed by:
`float((this->width * this->width) * 0.25 * PI)`
*
with width set to nozzle_diameter if bridge_fflow_ratio is at 100%

CCS86, 2021-01-23:
Well, that explains it. 

Have you done any comparison testing between this cylindrical extrusion flow calc, and one calculated like a normal extrusion?

robthide37, 2021-01-25:
I am not talking about the layer height. How can you make a .2 layer height
in the middle of the air ? it is impossible if you have a .4 nozzle .You
are also misunderstanding the setting . the 50 percent and hundred percent
refers to the amount of the line that is unsupported before switching to
the bridge flow values it is not a percentage of the bridge flow. If you
hover over the value it will list a short description verifying this for
you so its not just taking my word for it.

CCS86, 2021-01-25:
I am not misunderstanding the setting. 

I am talking about the bridge flow ratio. You are talking about bridge flow overhang threshold.

robthide37, 2021-01-25:
> I am not misunderstanding the setting.
> 
> I am talking about the bridge flow ratio. You are talking about bridge flow overhang threshold.

Sorry you are correct i understood the original post but for some reason i assumed you were referring to the trigger value when you responded to me i am not really sure what made me think of that i just read the email and responded but now i am on the actual website and can see the full thread . My only guess on the 50 percent is that that would work for your situation at .2 layer height and prob 25 percent would work at .1 and 75 at .3. I am going to try these values as now you have me wondering . Also btw just because theoretically it should be the case of using full nozzle diameter for extrusion size does not mean real world its ideal and i don't even use bridge flow for overhangs at all because of this but i could see situations where they both can apply . EX if you have a very cohesive and viscous filament like petg it could prob work well with lower values but then abs or something that flows real well i would think needs the full value to prevent droop "just an uneducated guess "

CCS86, 2021-01-25:
I did a bunch more testing on this and found surprising results. 

Basically, I set up a 50mm bridge span (true bridge) and repeated it varying different parameters (bridge flow, speed, cooling).

I actually got good results at very high bridge flow ratio settings. Working from bridge flow ratio between 60%-100% I preferred the result at 100%. I then went up and beyond, and still got good results at bridge flow at 135%. I actually had to uncap the volumetric flow limits to go that high. This is all with PLA.

There was definitely some amount of randomness in getting some droopy beads. I think most of the time it is caused by the interaction between the last bead and the current bead. I can't help but wonder if results would be better with modified bridge line spacing to ensure they don't touch. Or even doing every other bead first, then coming back to fill in between.

closed 2021-01-27 by CCS86

supermerill, 2021-06-02:
>  I then went up and beyond, and still got good results at bridge flow at 135%. I actually had to uncap the volumetric flow limits to go that high. This is all with PLA.

maybe a not-that-melted core in the extrusion help with bridges?