supermerill/SuperSlicer #881: CTRL+I opens a new SuperSlicer window in Linux Mint in versions 2.3.x

Rickerdo, 2021-01-14:
### Version
2.3.x

### Operating system type + version
Linux Mint 20.1

### 3D printer brand / version + firmware version (if known)
Ender 3 Pro/Klipper

### Behavior
When pressing CTRL+I a new SuperSlicer window opens.
Expected behavior is to open the import dialog box.
Issue does not exist in 2.2.53

neophyl, 2021-01-14:
This is the same as the alpha Prusa Slicer builds which the 2.3 builds are built on.  I remember seeing the issue open on their github.  What I cant recall is if they changed the behaviour for the proper 2.3 release.

Rickerdo, 2021-01-14:
Having used the RC candidates and the final v2.3 release of PrusaSlicer (on the same workstation), I can say with certainty that this issue does not exist in any of the 2.3 versions of PrusaSlicer. While I see where you're coming from, the issue does exist with SuperSlicer, so I would consider it something that needs to be fixed within SuperSlicer.

neophyl, 2021-01-14:
Ok then how do you explain this issue ?  https://github.com/prusa3d/PrusaSlicer/issues/4711

Rickerdo, 2021-01-14:
I can't explain it, nor do I care to. I've stated my experience with previous 2.3 releases of PS along with my current experience with the latest 2.3 release of PS. Perhaps I jumped on the PS boat after it was fixed? Who knows!? I don't expect this to become a pissing match, it's a request to fix a bug. The point here is that the expected behavior is broken. Can we focus on that?

neophyl, 2021-01-14:
If you would care to read the linked issue you would realise that you did 'jump on the boat' with the PS releases after the bug was fixed.  
As such when a new version of Super Slicer is released that is based on the Prusa Slicer 2.3 proper release the issue wont be there.  So Merill has nothing to fix.  There is no point spending any time something that will be fixed when the next merge is done.

Basically anyone reporting ANY issue on the non stable 2.3.55 versions of Super Slicer is wasting their time.  If the problem is still there on the NEXT release then by all means create an issue to be looked at.

Rickerdo, 2021-01-14:
Who pissed in your Cheerios this morning!? I did read the link you provided. No need for the second round of condescending remarks. If I understood any of this I would not have opened the issue in the first place. I'm simply an end user who knows nothing about any of which you stated, nor do I care. I saw a problem and reported it thinking it might be of help (in a very insignificant way). My bad! Had you actually taken a few seconds to step back and explain what you just stated (sans attitude) perhaps this thread would not have gone beyond your initial response and I would have moved along silently waiting for another release of SS. I get the point now.

neophyl, 2021-01-14:
I had to look up what Cheerios were.  Guess you do learn something new every day - thanks.  I'm a software tester by profession, we have our sense of humour surgically removed.

supermerill, 2021-01-23:
As the next version is merged with prusaslicer 2.3, this issue will be fixed by itself

thanks @neophyl for checking it.

closed 2021-06-04 by supermerill