supermerill/SuperSlicer #882: Volumetric autospeed does not respect the linear speed limit

zbrozek, 2021-01-14:
### Version
2.3.55.2

### Operating system type + version
Windows 10 Pro build 19042

### Behavior
1) Set the various linear speed limits to zero (e.g., auto).
2) Set the maximum linear speed to something finite (e.g., 120 mm/s).
3) Set the volumetric autospeed to something finite (e.g., 15 mm^3/s).
4) Slice a model.
5) Look at the linear speed in gcode visualizer and note that it exceeds (often enormously) the maximum linear speed setting.
[image]

#### Project File (.3MF) where problem occurs
[demo.zip](https://github.com/supermerill/SuperSlicer/files/5815795/demo.zip)

supermerill, 2021-01-18:
There is  two bugs here.

1) wrong speed for thinwalls & gapfill when autospeed.
2) when setting gapfill speed from somethign to 0, it doesn't recompute the perimeter, and so keep the gapfill.

I'm going to remove the "gapfill 0 speed deactivate the gapfill".
The new speed for gapfill/thinwall when autospeed will be set to the speed of a 200% width extrusion (if you have some opinions about that, tell me).

zbrozek, 2021-01-19:
Both of those sound good to me - thank you!

closed 2021-06-04 by supermerill

SirStigII, 2021-06-11:
I think I may have noticed a related issue in the latest stable release (2.3.56.4b), if speeds are set to auto like in the above situation, but the outer and top layers are set to a percentage (I was using 50%) they will slice at the max print speed, instead of using the correct percentage. setting the perimeter speed and solid speed to 120mm/s in this case will correctly use the 50% value (60mm/s)

supermerill, 2021-06-11:
> I think I may have noticed a related issue in the latest stable release (2.3.56.4b), if speeds are set to auto like in the above situation, but the outer and top layers are set to a percentage (I [...]

Give your project file to help me reproduce the problem.

SirStigII, 2021-06-11:
sorry, saw this one was only recently closed and figured it might be a connected issue. I'll upload the project file in a new issue.