supermerill/SuperSlicer #883: Crashes when opening printer settings using old configuration

NovaViper, 2021-01-16:
### Version
2.3.55.4

### Operating system type + version
Manjaro KDE v20.2.1

### 3D printer brand / version + firmware version (if known)
Creality Ender 3 + BTT SKR Pro v1.1 + Marlin bugfix 2.0.0

### Behavior
Whenever I go and open up the Printer settings in Super Slicer, it just immediately crashes.
* _Expected Results_
It should open the printer settings

* _Actual Results_
It just instantly closes, here is my configuration snapshot: [20210116T021544Z.zip](https://github.com/supermerill/SuperSlicer/files/5823692/20210116T021544Z.zip)

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`File -> Save` / `Save Project`)_

PePa-444, 2021-01-16:
The same here, windows 10 (20H2).

uepsie, 2021-01-16:
check #870

NovaViper, 2021-01-16:
> check #870

I don't have a called resources/ui_layout/printer_fff.ui . In fact I'm not even sure where the program gets installed to since im using an AUR (Arch User Repo) to install Super Slicer, this is the [aur im using](https://aur.archlinux.org/packages/superslicer-git/)

supermerill, 2021-01-17:
can't reproduce in the latest dev build (with my settiungs replaced by yours), on windows.
So it should be fixedin the next release.

khumarahn, 2021-01-17:
> > check #870
> 
> I don't have a called resources/ui_layout/printer_fff.ui . In fact I'm not even sure where the program gets installed to since im using an AUR (Arch User Repo) to install Super Slicer, this is the [aur im using](https://aur.archlinux.org/packages/superslicer-git/)

The file is `/usr/share/SuperSlicer/ui_layout/printer_fff.ui`. @supermerill I have this problem on linux with the appimage from https://github.com/supermerill/SuperSlicer/actions/runs/475680461 (or I think it is the same problem). Compiling from AUR and editing the file as in #870 helped.

NovaViper, 2021-01-18:
@khumarahn That fixed the issue for me, thanks!