supermerill/SuperSlicer #885: Slicer not responding on open

emaayan, 2021-01-16:
### Version
2.3.55.2 

### Operating system type + version
windows 10

### 3D printer brand / version + firmware version (if known)
Tevo Black widow 

### Behavior
slicer hangs when double clicking on superSlicer.exe
flash screen appears with prepearing settings tab.
clicking on the slicer turns into black (and title goes non responding)

soysnorre, 2021-05-09:
Hi, 
I have the same problem on windows 10 pro with last version 2.3.55.5.
I have overwritten all files and disable firewall to see what happen and the same. It's not respond i the same stage. I can look even tabs and etc, but It looks like table not charging.
Some idea??... thanks in advance

supermerill, 2021-05-20:
No ideas... maybe a deadlock bug, i've fixed some in the last month.