supermerill/SuperSlicer #887: Thumbnail/Gcode Preview for Slic3r/PrusaSlicer/SuperSlicer for BigTreeTech TFT Screens

sarvenn, 2021-01-18:
Hi,
Thumbnails generated by Slicer could not be viewed in BTT screens. Codes generated by slicer is not applicable for BTT TFT screens.

In BTT touch screen repository there is a custom solution for that. Please see the link below.
Could this solution be integrated to SuperSlicer?

#https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1238

supermerill, 2021-01-18:
I almost understand what the format is.

But can you link here a gcode file from SuperSlicer (or PS) with & without the script?

effgarces, 2021-01-18:
Here you go, a calibration cube sliced with superslicer (by the way thanks for superslicer, it's awesome), with (_btt.g file) and without the script.
[xyzCalibration_cube.zip](https://github.com/supermerill/SuperSlicer/files/5829999/xyzCalibration_cube.zip)

sarvenn, 2021-01-19:
I also plan to try superslicer. I started Cura since last year, then switched to PS a month ago. Actually Cura had more features but PS seems to be more fast (program itself and printing times). I would like to try SuperSlicer as well. I wish SS will integrate a solution for BTT TFT screen compatible gcode preview.

sarvenn, 2021-07-21:
@supermerill hi. Btt tft fw is now compatible with prusaslicer's thumbnail generation aporoach.

effgarces, 2021-07-21:
True, but not all of the TFT's support it, due to memory issues, so there's value in having this feature.

sarvenn, 2021-07-26:
> True, but not all of the TFT's support it, due to memory issues, so there's value in having this feature.

Great to hear that. Thnx.

sarvenn, 2022-02-15:
Somehow I closed this issue, I don't know how but it's better to have a look again here @supermerill .
If you find it's too hard to integrate this solution to SS, I can close the issue upon your request.

sarvenn, 2022-02-15:
I know that PS developers are not interested in.

supermerill, 2022-02-16:
I just merge the JPG / QOI thumbnail from prusa, adding that one.

supermerill, 2022-02-16:
Can someone test if you can see a yellow cube on white background?
Just copy-paste this on the start of a gcode file:
```
; generated by SuperSlicer 2.4 on 2022-02-15 at 22:56:35 UTC

;
; thumbnail_BIQU begin 50x50 10100
;00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000008401080000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
;0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000031a194848c44420100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
;00000000000000000000000000000000000000000000000000000000000000000000000000000000000000005ac294849484948494846b630840000000000000000000000000000000000000000000000000000000000000000000000000000000000000
;0000000000000000000000000000000000000000000000000000000000000000000000000000000018c084039484948494849484948494848c44296100000000000000000000000000000000000000000000000000000000000000000000000000000000
;00000000000000000000000000000000000000000000000000000000000000000000000000004a42948494849484948494849484948494849484948452820840000000000000000000000000000000000000000000000000000000000000000000000000
;0000000000000000000000000000000000000000000000000000000000000000000018c073a394849484948494849484948494849484948494849484948473a318c000000000000000000000000000000000000000000000000000000000000000000000
;000000000000000000000000000000000000000000000000000000000000000042019484948494849484948494849484948494849484948494849484948494848c4442010000000000000000000000000000000000000000000000000000000000000000
;00000000000000000000000000000000000000000000000000000000084063229484948494849484948494849484948494849484948494849484948494849484948494846b63108000000000000000000000000000000000000000000000000000000000
;000000000000000000000000000000000000000000000000000018c08c44948494849484948494849484948494849484948494849484948494849484948494849484948494848c4429610000000000000000000000000000000000000000000000000000
;00000000000000000000000000000000000000000000000052829484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494845282084000000000000000000000000000000000000000000000
;000000000000000000000000000000000000000018c073a39484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948473a318c00000000000000000000000000000000000000000
;0000000000000000000000000000000000004201948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494848c444201084000000000000000000000000000000000
;000000000000000000000000000008406322948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494846b6310800000000000000000000000000000
;0000000000000000000000002961948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494848c4431a1000000000000000000000000
;000000000000000000005ac2948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494845ac200200000000000000000
;0000000000000000108073838c649484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484632310800000000000000000
;000000000000000010805ae26b4384039484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484842352a24a6210800000000000000000
;000000000000000010805ae25ae263027bc38c64948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948473834a624a624a6210800000000000000000
;000000000000000010805ae25ae25ae25ae26b438c44948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948484235ac24a624a624a624a6210800000000000000000
;000000000000000010805ae25ae25ae25ae25ae263027bc39484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948473a34a624a624a624a624a624a6210800000000000000000
;000000000000000010805ae25ae25ae25ae25ae25ae25ae26b438c6494849484948494849484948494849484948494849484948494849484948494849484948494849484948494848c6463234a624a624a624a624a624a624a6210800000000000000000
;000000000000000010805ae25ae25ae25ae25ae25ae25ae25ae26302840394849484948494849484948494849484948494849484948494849484948494849484948494849484842352a24a624a624a624a624a624a624a624a6210800000000000000000
;000000000000000010805ae25ae25ae25ae25ae25ae25ae25ae25ae25ae273a38c64948494849484948494849484948494849484948494849484948494849484948494846b634a624a624a624a624a624a624a624a624a624a6210800000000000000000
;000000000000000010805ae25ae25ae25ae25ae25ae25ae25ae25ae25ae25ae26b438c449484948494849484948494849484948494849484948494849484948484235ac24a624a624a624a624a624a624a624a624a624a624a6210800000000000000000
;000000000000000010805ae25ae25ae25ae25ae25ae25ae25ae25ae25ae25ae25ae263027bc38c649484948494849484948494849484948494849484948473a34a624a624a624a624a624a624a624a624a624a624a624a624a6210800000000000000000
;000000000000000010805ae25ae25ae25ae25ae25ae25ae25ae25ae25ae25ae25ae25ae25ae26b438c44948494849484948494849484948494848c6463024a624a624a624a624a624a624a624a624a624a624a624a624a624a6210800000000000000000
[... cut ...]
;000000000000000000000000000000000000000000000000000000000000000000000000000000000000002031a15ae25ae24a624a6229610000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
;0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000010a052824a4218e000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
;00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000200840000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
; thumbnail_BIQU end
;
; 
```

sarvenn, 2022-02-16:
Oh btw, this is another PR for thumbnail. https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/2292

supermerill, 2022-02-16:
> Send me a gcode file, I'll try it. May be it requires 95x95 pixel icon on my TFT35.

here: [Shape-Box_bqu.zip](https://github.com/supermerill/SuperSlicer/files/8081398/Shape-Box_bqu.zip)

sarvenn, 2022-02-16:
No, not working 😐
This is a working example on my TFT35, compiled with PS 2.4.1beta and post processed with @effgarces 's script .(https://github.com/effgarces/Biqu-Thumbnail-Generator)

[xyzCalibration_cube_0.2mm_39m_PLA_Microzey_Turuncu_btt.zip](https://github.com/supermerill/SuperSlicer/files/8081792/xyzCalibration_cube_0.2mm_39m_PLA_Microzey_Turuncu_btt.zip)

sarvenn, 2022-02-16:
No, not working still.
There is another weird thing, if I open one of the working file with VSCode and do something, revert it and save the file, like add a space and then backspace and save the file. Then thumbnail is not visiblity anymore. When I do this with notepad, there is no problem.
When I open the file with notepad file consists of chineese letters, but I can open with VSCode. But I'm suspicious about editing a file with VSCode due to the concern above.
[image]

sarvenn, 2022-02-16:
[notepad.zip](https://github.com/supermerill/SuperSlicer/files/8081976/notepad.zip)
These 3 files are exactly same.
Original is the one I sent to you previously, xyz cube.
VSCode is opened with VSCode and edited (just a space and backspace move and save)
Notepad is opened with notepad and edited ((just a space and backspace move and save)

Result: Original and Notepad thumbnails are still visible but VSCode is not visible.
May be this will attract @effgarces too.

I will try to open and edit with MS Word as well.

supermerill, 2022-02-16:
Maybe your app is messing with the charset & line ending.
Use notepad++ or sublimetext. You shouldn't see any utf character, but it so, set your encoding to utf-8.
Be sure you only have \n and no \r

I don't understand why my file doesn't work, and can't progress from here. So until someone can guide me on what's wrong, I stop dev on this.

supermerill, 2022-02-16:
Last try: I added the `; bigtree thumbnail end` at the end of the thumbnail
[Shape-Box_bqu.gcode.txt](https://github.com/supermerill/SuperSlicer/files/8082187/Shape-Box_bqu.gcode.txt)

sarvenn, 2022-02-16:
Sorry, it didn't work. :(
I hope some other BTT TFT and script users also try and send their feedback here.

effgarces, 2022-02-23:
I just took a look a the file Shape-Box_bqu.gcode.txt, I don't see anything wrong with it, but I can't test it on my TFT and maybe that the issue with sarvenn, the file only has the thumbnail for the 95x95 TFT, the TFT35, I have a TFT24 that requires the Thumbnail to be in 70x70 resolution, as far as I am aware each TFT just searches/loads their corresponding resolution.
According to Biqu this is resolution table:
- 70x70: TFT24 / TFT28
- 95x80: TFT43 / TFT50
- 95x95: TFT35
- 160x140: TFT70

Abyss8, 2022-03-17:
Hi i have SKR 1.4 Turbo / Marlin 2.0.9.3 / BTT TFT35 E3 BIGTREE_TFT35_V3.0_E3.27.x (THUMBNAIL_PARSER 2 )
Tried those files and here is my results:

> Here you go, a calibration cube sliced with superslicer (by the way thanks for superslicer, it's awesome), with (_btt.g file) and without the script. [xyzCalibration_cube.zip](https://github.com/supermerill/SuperSlicer/files/5829999/xyzCalibration_cube.zip)
[image]

Hi mate tried your file and here is my result:

> > Send me a gcode file, I'll try it. May be it requires 95x95 pixel icon on my TFT35.
> 
> here: [Shape-Box_bqu.zip](https://github.com/supermerill/SuperSlicer/files/8081398/Shape-Box_bqu.zip)

[image]

Abyss8, 2022-03-17:
> Can someone test if you can see a yellow cube on white background? Just copy-paste this on the start of a gcode file:
> 
> ```
> ; generated by SuperSlicer 2.4 on 2022-02-15 at 22:56:35 UTC
> 
> 
> ;
> ; thumbnail_BIQU begin 50x50 10100
> ;00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000008401080000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
> ;0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000031a194848c44420100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
> ;00000000000000000000000000000000000000000000000000000000000000000000000000000000000000005ac294849484948494846b630840000000000000000000000000000000000000000000000000000000000000000000000000000000000000
> ;0000000000000000000000000000000000000000000000000000000000000000000000000000000018c084039484948494849484948494848c44296100000000000000000000000000000000000000000000000000000000000000000000000000000000
> ;00000000000000000000000000000000000000000000000000000000000000000000000000004a42948494849484948494849484948494849484948452820840000000000000000000000000000000000000000000000000000000000000000000000000
> ;0000000000000000000000000000000000000000000000000000000000000000000018c073a394849484948494849484948494849484948494849484948473a318c000000000000000000000000000000000000000000000000000000000000000000000
> ;000000000000000000000000000000000000000000000000000000000000000042019484948494849484948494849484948494849484948494849484948494848c4442010000000000000000000000000000000000000000000000000000000000000000
> ;00000000000000000000000000000000000000000000000000000000084063229484948494849484948494849484948494849484948494849484948494849484948494846b63108000000000000000000000000000000000000000000000000000000000
> ;000000000000000000000000000000000000000000000000000018c08c44948494849484948494849484948494849484948494849484948494849484948494849484948494848c4429610000000000000000000000000000000000000000000000000000
> ;00000000000000000000000000000000000000000000000052829484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494845282084000000000000000000000000000000000000000000000
> ;000000000000000000000000000000000000000018c073a39484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948473a318c00000000000000000000000000000000000000000
> ;0000000000000000000000000000000000004201948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494848c444201084000000000000000000000000000000000
> ;000000000000000000000000000008406322948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494846b6310800000000000000000000000000000
> ;0000000000000000000000002961948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494848c4431a1000000000000000000000000
> ;000000000000000000005ac2948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494845ac200200000000000000000
> ;0000000000000000108073838c649484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484632310800000000000000000
> ;000000000000000010805ae26b4384039484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484842352a24a6210800000000000000000
> ;000000000000000010805ae25ae263027bc38c64948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948473834a624a624a6210800000000000000000
> ;000000000000000010805ae25ae25ae25ae26b438c44948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948484235ac24a624a624a624a6210800000000000000000
> ;000000000000000010805ae25ae25ae25ae25ae263027bc39484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948494849484948473a34a624a624a624a624a624a6210800000000000000000
> ;000000000000000010805ae25ae25ae25ae25ae25ae25ae26b438c6494849484948494849484948494849484948494849484948494849484948494849484948494849484948494848c6463234a624a624a624a624a624a624a6210800000000000000000
> ;000000000000000010805ae25ae25ae25ae25ae25ae25ae25ae26302840394849484948494849484948494849484948494849484948494849484948494849484948494849484842352a24a624a624a624a624a624a624a624a6210800000000000000000
> ;000000000000000010805ae25ae25ae25ae25ae25ae25ae25ae25ae25ae273a38c64948494849484948494849484948494849484948494849484948494849484948494846b634a624a624a624a624a624a624a624a624a624a6210800000000000000000
> ;000000000000000010805ae25ae25ae25ae25ae25ae25ae25ae25ae25ae25ae26b438c449484948494849484948494849484948494849484948494849484948484235ac24a624a624a624a624a624a624a624a624a624a624a6210800000000000000000
> ;000000000000000010805ae25ae25ae25ae25ae25ae25ae25ae25ae25ae25ae25ae263027bc38c649484948494849484948494849484948494849484948473a34a624a624a624a624a624a624a624a624a624a624a624a624a6210800000000000000000
[... cut ...]
> ;00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000200840000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
> ; thumbnail_BIQU end
> ;
> ; 
> ```
> 
> edit: I spot an error: it misses the size header

Don't works.

> Last try: I added the `; bigtree thumbnail end` at the end of the thumbnail [Shape-Box_bqu.gcode.txt](https://github.com/supermerill/SuperSlicer/files/8082187/Shape-Box_bqu.gcode.txt)

Don't works.

Abyss8, 2022-03-17:
Tried editing file "xyzCalibration_cube_btt.g" from  "xyzCalibration_cube.zip"  (effgarces commented on 18 Jan 2021)

And for working and see icon file need 3+ cubes:
[image]
[Editing xyzCalibration_cube_btt_g.zip](https://github.com/supermerill/SuperSlicer/files/8293620/Editing.xyzCalibration_cube_btt_g.zip)

supermerill, 2022-03-18:
So the solution is to create 3 thumbnails?
have to add another thumbnail size field then.

Abyss8, 2022-03-18:
It turns out that it is.
I am willing to help, so feel free and i will test :)
SuperSlicer should have such a facility for users of these BTT TFT screens.
Thanks for fast response.

supermerill, 2022-03-18:
if you change in your printer_fff.ui
```
group:Thumbnails
	line:Size for Gcode
		setting:id$0:label$Small:thumbnails
		setting:id$1:label$Big:thumbnails
	end_line
```
to
```
group:Thumbnails
	line:Size for Gcode
		setting:id$0:label$Small:thumbnails
		setting:id$1:label$Medium:thumbnails
		setting:id$2:label$Big:thumbnails
	end_line
```
then you'll have 3 size and can export 3 good sizes.
but I guess that from the gcode, i have to avoid outputting 
```
; bigtree thumbnail end
; thumbnail_BIQU end
;

;
; thumbnail_BIQU begin 40x40 6480
```
between each one?

(i used 20x20, 40x40 and 80x80, i don't know if it needs exactly the right size or not)
[Shape-Box_guess_good.gcode.txt](https://github.com/supermerill/SuperSlicer/files/8303401/Shape-Box_guess_good.gcode.txt)
[Shape-Box_guess_bad.gcode.txt](https://github.com/supermerill/SuperSlicer/files/8303402/Shape-Box_guess_bad.gcode.txt)

Abyss8, 2022-03-18:
I checked both of your gcode and unfortunately you can't see the icon:
[image]

I tried to edit and delete the lines of text in front of gcode and at the end of this one myself.
I reduce the spacing as in the file that works for me, but unfortunately don't works.
[SuperSlicer TFT tests.zip](https://github.com/supermerill/SuperSlicer/files/8303692/SuperSlicer.TFT.tests.zip)

supermerill, 2022-03-18:
here with the same dimension as the one you sent to me (40x40, 95x80, 95x95)
[Shape-Box_guess_bad.gcode.txt](https://github.com/supermerill/SuperSlicer/files/8304572/Shape-Box_guess_bad.gcode.txt)
[Shape-Box_guess_good.gcode.txt](https://github.com/supermerill/SuperSlicer/files/8304568/Shape-Box_guess_good.gcode.txt)
[Shape-Box_guess_dunno.gcode.txt](https://github.com/supermerill/SuperSlicer/files/8304566/Shape-Box_guess_dunno.gcode.txt)

Abyss8, 2022-03-18:
> > Send me a gcode file, I'll try it. May be it requires 95x95 pixel icon on my TFT35.
> 
> here: [Shape-Box_bqu.zip](https://github.com/supermerill/SuperSlicer/files/8081398/Shape-Box_bqu.zip)

I forgot to check this file but don't work.
Modified to first line is: ;005f005f  = nothing
Modified to:
; bigtree thumbnail end
; thumbnail_BIQU end
and
; thumbnail_BIQU end
; bigtree thumbnail end
= nothing

> here with the same dimension as the one you sent to me (40x40, 95x80, 95x95) [Shape-Box_guess_bad.gcode.txt](https://github.com/supermerill/SuperSlicer/files/8304572/Shape-Box_guess_bad.gcode.txt) [...]

Tested all files and diddn't see the icon :(

supermerill, 2022-03-18:
out of clues then.

Abyss8, 2022-03-18:
We have one file generated from

> Here you go, a calibration cube sliced with superslicer (by the way thanks for superslicer, it's awesome), with (_btt.g file) and without the script. [xyzCalibration_cube.zip](https://github.com/supermerill/SuperSlicer/files/5829999/xyzCalibration_cube.zip)
xyzCalibration_cube_btt.gcode
that works.
I don't used yet that script but file generated from that works.
So i think we need to see whats exacly do script and recreate it.

Abyss8, 2022-03-18:
Or is someting in that file we ignoring but that makes a diffrence.

supermerill, 2022-03-18:
i don't spot any significant  diff from my generated one.

Abyss8, 2022-03-18:
Ok little progress here:

[image]

Abyss8, 2022-03-18:
So i take your image from first line 1 to almost here: ; external perimeters extrusion width = 0.63mm
Copied
And Paste into scripted file from 1 to ; external perimeters extrusion width = 0.63mm
[image]
[Nowy folder2.zip](https://github.com/supermerill/SuperSlicer/files/8308066/Nowy.folder2.zip)

Picture from Shape-Box_guess_good do the job.
Something else must be causing the problem.

Tried: 
swap "object:" / deleted "plater:" / deleted "M486 T1" / deleted ; "Total objects to print: 1" / deleted "; thumbnail_BIQU end" / deleted ;"TYPE:Custom"
but from that nothing works.

Something is in that scripted file. If i swap only line from 1 to that on photo i see a icon.

Abyss8, 2022-03-19:
I tried to replace various lines of text, even those that should not cause a problem.
[image]

But I can't get to the difference in the file that causes the icon to be missing.

There is progress because your icon started to work better than nothing :)
As mentioned above, we copy the photo from your file "good"and paste them into the generated file with the script and the icon magically appears.

It doesn't match the line of text at the end at all gcoda "; thumbnails = "70x70,95x80,95x95"  and "; thumbnails = 0x0,200x200"
but it just works. I wasn't sure if the printer didn't read the end of the gcode so I started changing the lines of the text there too, but unfortunately I didn't get to it.

Have a look at it with your experienced eye, because I was just trying to change the lines of the text and the sequence among myself, but I don't know much about programming and apparently I don't see anything else.

supermerill, 2022-03-23:
@Abyss8 
If i understand, it doesn't work but if you copy the thumbnail in a file that works, it's working.

Can you send me a project that creates a gcode file for your printer, the gcode file and the gcode file after passing through the script?
i'll see if I can use that project to create a better gcode file, or check where the differenced are.

Abyss8, 2022-03-23:
@supermerill This is not mine gcode.
@effgarces Created it:

> Here you go, a calibration cube sliced with superslicer (by the way thanks for superslicer, it's awesome), with (_btt.g file) and without the script. [xyzCalibration_cube.zip](https://github.com/supermerill/SuperSlicer/files/5829999/xyzCalibration_cube.zip)

I have problem with Python on my PC when i try any script.
Maybe  i will try on my laptop or VirtualBox but i think you can do it better than me.
https://github.com/effgarces/Biqu-Thumbnail-Generator
@effgarces Mate can you help?

effgarces, 2022-03-24:
Hi, I've been away on work, so no chance of contributing.  The script is not my work, as you can clearly see on the repo, it was shared over at https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1238, I just modded it slightly to fit my needs and packaged it in an executable, and also modified it a little to work with prusaslicer 2.4
I will see if during the weekend I can get back into this and try to contribute.

effgarces, 2022-03-29:
I've been testing the provided files, but have also come up empty, I'm just not seeing what could be stopping the preview from working. 
Here is the files that you requested @supermerill , a project for my printer (I have the TFT24), the gcode generated and the gcode after passing the script:
[xyzCalibration_cube.zip](https://github.com/supermerill/SuperSlicer/files/8369828/xyzCalibration_cube.zip)
Hopefully these will be helpful.

Abyss8, 2022-08-14:
Recently I installed the Arachne version and I am positively surprised :) A lot of configuration options, I also noticed that the seam visibility was added, which I missed a lot, and snug supports, extra 👍 

But I also noticed that the options for selecting Thumbnails, Biqu,.. were added. I tested all options on 95x95 and 'Bed on thumbnail, Print at the end' but unfortunately no option is supported by TFT. 
My guess is that this is not a priority and there are more important things to introduce, but I would like to ask if there is any plan to implement support for these BTT displays, or is work on it suspended at the moment and it is not known if it will be. 

There is no need to rush because there are more interesting options available and worth more, but it would be nice to add it.
If I can help with the tests or anything else, just write. 
Thanks for SuperSlicer, I'm still crawling at the moment and getting to know him because he is very advanced and has a lot of options.

adripo, 2022-10-30:
Hi everybody, I tried to create a PR to generate MKS TFT thumbnails, but I don't have much experience with c++. Can anybody check and help me? I added more details in the PR description https://github.com/supermerill/SuperSlicer/pull/3387 . Thank you!

Lucas-AR97, 2022-12-23:
Hi everybody, I can to create my thumbnails with [png2mks](https://github.com/slawkis/png2mks) for my Hellbot Magna SE Pro - MKS FTF35.
Follow this guide and you will be able to generate your thumbnails without problems.
Hopefully this will be helpful.

Abyss8, 2023-06-08:
> Hi everybody, I can to create my thumbnails with [png2mks](https://github.com/slawkis/png2mks) for my Hellbot Magna SE Pro - MKS FTF35. Follow this [guide [...]

Can you post gcode generated by that program? (cube.stl)

Abyss8, 2023-06-09:
@supermerill 

Ok, I have something interesting, I tried to approach this topic again and try to solve this puzzle.
I made interesting observations, a few things became clear and I'm even able to "create" gcode that will correctly display the icon but it's a bit of work :D

So yes, the file:
1# DZIALA_good.gcode (WORKS), 
[image]
#############################################################

2# TEZDZIALANOWELOGO_DZIALA_good.gcode (WORKS), 
[image]
############################################################

3# NIE_DZIALA_NOWYGCODE_6h13mssss_0.2mmPETG0.4n_.gcode (NOT WORKING).
[image]
############################################################

1# This is a file generated sometime there and it works, let's call it MAGIC
2# This is a MAGIC file but a new generated icon is added to it
3# This is a new file but all characters (CTRL+A) (CTRL+C) (CTRL+V) from the MAGIC file have been pasted into it, so it's the same but it DOES NOT WORK....
[image]
COMPARED 
2# [MAGIC FILE + NEW ICON] AND  3# [NEW GENERATED GCODE AND ALL DATA FROM MAGIC FILE+NEW ICON]
MATCH! AND NEW FILE NOT WORKING
###########################################################

I created a very professional representation of this magical thing:
[image]

What is this magic?

I am able to create a gcode with a new icon, but you have to cut something here and paste something there
[image]

Do you understand any of this?
[files.zip](https://github.com/supermerill/SuperSlicer/files/11709327/files.zip)

discip, 2023-10-13:
@Abyss8
Here is what actually makes the files work for the BTT TFTs.
https://github.com/SoftFever/OrcaSlicer/issues/1894#issuecomment-1760509410

Just so you know. 😊