supermerill/SuperSlicer #888: Window keeps raising itself on gaining focus

VanessaE, 2021-01-18:
### Version

git commit b6d3143c868c4713f71c20e64d4cc631c93e8400

### Operating system type + version

Debian Bullseye/testing, XFCE 4.16

### 3D printer brand / version + firmware version (if known)

N/A

### Behavior

As is common on Linux systems, I have my desktop set to focus-follows-mouse, don't raise windows on clicking, and don't raise on gaining focus.  When I need to, I intentionally raise or bury a window with ALT plus one of the mouse buttons.

However, SuperSlicer frequently raises itself after gaining and then losing focus, after a very short delay, even if I do nothing other than slide the mouse pointer across it while aiming for something else sitting on top of it.

You can reproduce this pretty simply:  Open SuperSlicer, maximize it, then open a terminal or some other simple thing, and place it above the SuperSlicer window.  Move your mouse back and forth between the two, causing them to repeatedly gain and lose focus, then stop moving, leaving the mouse pointer on the terminal window.  After a moment, SuperSlicer will raise itself.

This only seems to happen if the window is maximized.

This also affects Prusaslicer (https://github.com/prusa3d/PrusaSlicer/issues/5620), but not any other application on my system.

This is a relatively new phenomenon -- I cannot be 100% certain, but I think it started some time between commits 996a42f2f95117805470b62f1504a86ae6ca33a0 and b6d3143c868c4713f71c20e64d4cc631c93e8400.  The former is the commit I was using last before updating today, and I could swear that I wasn't able to reproduce this phenomenon before today (i.e. when I filed the Prusaslicer bug).

#### Project File (.3MF) where problem occurs
N/A

supermerill, 2021-01-18:
seems something pulled from the PS2.3 code.

VanessaE, 2021-01-30:
Oh my G*D, someone PLEASE find the cause of this and fix it.  It is by far the most annoying thing ever to have a program pop up in my face repeatedly after I bury/hide it.

RealDeuce, 2021-09-01:
On my XFCE desktop (which uses click-to-focus), this occurs only when I use ALT-Tab to switch focus, it does not occur when I use the mouse to change focus.

Highly annoying though.

RealDeuce, 2021-09-01:
Looks like it was introduced with this commit: 20f5b7a7cd

RealDeuce, 2021-09-01:
Specifically, adding CallAfter() appears to be the cause... calling them immediately rather than using CallAfter() appears to work.

RealDeuce, 2021-09-02:
Pr submitted upstream