supermerill/SuperSlicer #893: A way to improve soluble support material adhesion by base type interface layers under this materials.

spiky2021, 2021-01-19:
### Version
2.30

### Operating system type + version
Win10

### 3D printer brand / version + firmware version (if known)
Anycubic I3 Mega S, Craftbot Flow IDEX XL

### Behavior
At the moment soluble support material adhesion is weak due to sparse support layers under soluble support layers.
[image]
Now I added two modified methods to the SupportMaterial Class (.cpp and .hpp attached).
[new_SupportMaterial_Class_files.zip](https://github.com/supermerill/SuperSlicer/files/5834406/new_SupportMaterial_Class_files.zip)
[image]
This adds two base type interface layers to the support structure, in case the extruders are different and soluble support is choosen.
Now soluble support adhesion is much stronger.
If you are willing, it might be an easy task to add a GUI input for the number of base interface layers.
Though I believe two layers are reasonable for most situations.

By the way, I looks like there is a bug in the original generate_interfaces method. The " idx_top_contact_first" is searched under the condition "layer->print_z >= intermediate_layer.print_z", but it should be "layer->bottom_z >= intermediate_layer.print_z" to avoid excess layers by swapping a layer too later to the upper top contact. In other words, compared to the bottom contact code it should be mirrow symmetric.

I am going this way, since I am new to the PrusaSlicer framework and GitHub (1 Week experience). But I hope you will bring it to the base branch.
#### Project File (.3MF) where problem occurs

spiky2021, 2021-01-21:
Added Class files included a bug. Now a commenting was removed and a fix for excess layers was done. The index for the top_z/bottom_z discriminators uses the interface layer number. Since indexes usually start at zero, the number needed to be reduced by one.
Original behaviour for two interface layers requested (actual PS 2.3.0 Version as well):
[image]
And now:
[image]
Support_with_base_support_interfaces_corrected
Supports are generated by painting. One can easily see the two base type support, as well.
New Class files code:
[new_SupportMaterial_Class_files.zip](https://github.com/supermerill/SuperSlicer/files/5848490/new_SupportMaterial_Class_files.zip)
A picture of a first print:
[image]
Materials: Extruder Greentec Pro and Polymaker Polydissolve S1.
Stringing and artifacts are generated by undry Polydissolve. Meaning the stiffness against the feeder and the viscosity was wrong. Without base interface layers I couldn't achieve prints with such tiny surfaces like 2x10mm, even with dry filament.

Done with a Craftbot Flow IDEX XL. No wipe tower, just used the wipers of the printer. It reduced the number of head preparation and therefore time and material costs by about 50%, even for this narrow top to bottom distances.
Supports were generated by painting and XY-Distance to zero, to achieve a support touching to tilted surfaces.
Here maybe soluble interface should be inflated to achieve a touching without this workaround.
For simple interfaces structures now even soluble support prints with single head printers by adding a pause or material change are possible.

The tiny dot of polydissolve on the bed is introduced to overcome a bug of PS for the multi extruder toolchange.
The generated gcode always introduces a normal enter retract/unretract for the first use of the head, instead of using a retraction procedure defined in a custom toolchange code. It leads to missing or overshoot material in the nozel, that leaves artifacts on the object and even worse deminishes adhesion.

ozolli, 2021-03-30:
Hello all,
This is a very useful feature indeed ! Support material may be 2 to 4 times the main material price and using the same main material all the way up to the supports top speeds up the prints by a lot !
Will it be included in a future realease ?
Many thanks in advance for your answer.