supermerill/SuperSlicer #896: Problem: Why can i not add the Ender 3 BL Touch

3DNichtVersteher, 2021-01-19:
### Version
2.3.55.4

### Operating system type + version
Windows 10 64bit_

### 3D printer brand / version + firmware version (if known)
Ender 3 Pro - Last Klipper Firmware

### Behavior

Hi,
Hello, I can't add the Ender 3 BL Touch in the SuperSlicer? He always tells me that no standard filament has been selected. The Prusa Slicer is all about it. What could be the reason for this? Greeting. Translated with Google.

ardyesp, 2021-01-23:
Hello, BLTouch is to be configured in the firmware (klipper). Slicer does not care about the BLtouch and will work without it. Attach your project file from superslicer and prusaslicer and we can check configuration.

a7hybnj2, 2021-06-13:
I just installed `SuperSlicer_2.3.56.5_macos_210612.dmg` which was just built and this is still an issue. Has it not made it to master yet? I will just follow your fix for now.

Thanks

supermerill, 2021-06-13:
maybe the commit was lost when moving setting to their new home, i'll check.

supermerill, 2021-06-21:
yes, the fix isn't here yet.

closed 2021-07-10 by supermerill