supermerill/SuperSlicer #898: Slicer is not honoring the Filaments values, taking it from tool changer

Brokengeiger, 2021-01-20:
### Version
SuperSlicer
Version:   2.3.55.2
Build:     SuperSlicer_2.3.55.2

### Operating system type + version
Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS Mojave Version 10.14.6 (Build 18G7016)
Total RAM size [MB]: 17,180MB
OpenGL installation
GL version:   2.1 INTEL-12.10.25
Vendor:       Intel Inc.
Renderer:     Intel(R) Iris(TM) Plus Graphics 655
GLSL version: 1.20

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

Rigidbot original --> But heavily modified dual extruder E3dV6 using Reliabuilds latest electronics for a v3.

### Behavior
* Setup super slicer for the 1st time
* configured for my printer
* added a 2nd extruder
* added in filament (Black PETG 250 C) 
* set both extruders to be the PETG
* generated bed leveling gcode
* attempted to print
* Temp raised to 200 
* attempted to print at 200

```
; filament_toolchange_temp = 200,200
; filament_type = other0;PLA
; filament_unload_time = 0,0
; filament_unloading_speed = 90,90
; filament_unloading_speed_start = 100,100
; filament_use_fast_skinnydip = 0,0
; filament_use_skinnydip = 0,0
; filament_vendor = (Unknown)
; filament_wipe_advanced_pigment = 0.5,0.5
; fill_angle = 45
; fill_angle_increment = 0
; fill_density = 18%
; fill_pattern = stars
; fill_smooth_distribution = 10%
; fill_smooth_width = 50%
; fill_top_flow_ratio = 100%
; first_layer_acceleration = 0
; first_layer_bed_temperature = 80,0
; first_layer_extrusion_width = 140%
; first_layer_flow_ratio = 100%
; first_layer_height = 75%
; first_layer_infill_speed = 30
; first_layer_size_compensation = 0
; first_layer_speed = 30
; first_layer_temperature = 200,200
```

Doesn't seem to be honoring the 250 I set for the filament. But it is pulling from the multitool section, upon changing this, the gcode still has references to 200, even though I set both extruder slots to 250. So it doesn't seem the filament extruder setting is ever honored and only the multitool section is used which is confusing.  

As a first time user for Prusa type slicers I'm not sure what to expect. But it does not feel like I'm getting the right values out of what I input. 

```
; filament_toolchange_temp = 250,200
; filament_type = other0;PLA
; filament_unload_time = 0,0
; filament_unloading_speed = 90,90
; filament_unloading_speed_start = 100,100
; filament_use_fast_skinnydip = 0,0
; filament_use_skinnydip = 0,0
; filament_vendor = (Unknown)
; filament_wipe_advanced_pigment = 0.5,0.5
; fill_angle = 45
; fill_angle_increment = 0
; fill_density = 18%
; fill_pattern = stars
; fill_smooth_distribution = 10%
; fill_smooth_width = 50%
; fill_top_flow_ratio = 100%
; first_layer_acceleration = 0
; first_layer_bed_temperature = 80,0
; first_layer_extrusion_width = 140%
; first_layer_flow_ratio = 100%
; first_layer_height = 75%
; first_layer_infill_speed = 30
; first_layer_size_compensation = 0
; first_layer_speed = 30
; first_layer_temperature = 250,200
```
[image]

[image]

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`File -> Save` / `Save Project`)_

[patch_bed_leveling.txt](https://github.com/supermerill/SuperSlicer/files/5839358/patch_bed_leveling.txt)

neophyl, 2021-01-20:
You need to attach a project 3mf file to have any hope of someone debugging the issue. Zip up the 3mf file to attach it here.

Brokengeiger, 2021-01-20:
I'll see if I can find the built into project bed leveling 3mf that comes with the project to submit it back to them. I did post the gcode, as that felt more relevant as to the results of debugging. 

Yeah here it is:
https://github.com/supermerill/SuperSlicer/tree/master/resources/calibration/bed_leveling

Brokengeiger, 2021-01-21:
I cannot imagine this helping much any more than the gcode as I mentioned it's a built in generated file, but alas _making the sure the power plug is in the wall and the computer_. attached. 
[patch_bed_leveling.3mf.zip](https://github.com/supermerill/SuperSlicer/files/5847202/patch_bed_leveling.3mf.zip)

supermerill, 2022-04-13:
seems fixed in 2.4, maybe also in 2.3.