supermerill/SuperSlicer #899: [BUG] One layer is missing while raft layers > 0

scott0122, 2021-01-20:
### Version
2.3.55.4, latest master code

### Operating system type + version
Win10.x64

### Behavior
One layer is missing while raft layers > 0

https://user-images.githubusercontent.com/39105828/105125850-5185d480-5b18-11eb-807a-d3b2de2f2950.mp4

liftbag, 2021-01-20:
Can you check if #685 is your issue?

scott0122, 2021-01-20:
@liftbag No. When I set Z-offset type to none, the missing layer are back.

scott0122, 2021-01-20:
@liftbag PrusaSlicer 2.3.0 has the same problem.

https://user-images.githubusercontent.com/39105828/105213139-4c14a280-5b89-11eb-88b9-12236567185b.mp4

liftbag, 2021-01-20:
> @liftbag No. When I set Z-offset type to none, the missing layer are back.

The problem is not very clear to me, given the information you provided, but I try to summarize, highlighting a behavior of SuperSlicer which, in my opinion, should be changed.

- I assume you are referring to the distance between the object and the raft.
- I also assume, in a completely hypothetical way, that you are using a 0.4 nozzle and a layer height of 0.2mm (this condition exactly matches your statement of a missing layer).

Choosing "None" as Z-offset type, is equivalent to using soluble supports, so there is no spacing between raft/supports and the object.
Choose anything else, it allows the spacing between raft/supports and object.

What in my opinion is not correct is that, only between object and raft, if you choose _From plane_ or _From filament_ as Z-offset type, the size of the nozzle is automatically inserted.
The distance indicated as Top Z-offset is added to the size of the nozzle.

For example, with the following settings...
[image]

[image]

[image]

...the distance between the object and the last raft layer will be 0.5mm, that is nozzle size + 0.1mm.
It is not possible in any way to reduce the distance determined by the nozzle, as negative values are not accepted.
It would be more useful, in my opinion, to be able to bring the object closer to the raft. In the slicing engine it would be more useful, and also understandable, to use the layer height in addition to the Top Z-offset, instead of the nozzle size.

If that's not your problem, I don't know how to reproduce it with the information you provided.

liftbag, 2021-01-20:
> @liftbag PrusaSlicer 2.3.0 has the same problem.
> 
>  2021-01-21_01-36-30.mp4

Ok. In the example with PrusaSlicer it is evident that the object loses a layer when soluble supports are not used. Let's see what supermerill thinks.

liftbag, 2021-01-20:
I can reproduce the issue with SuperSlicer too. Five layer with soluble supports against four layers when z-offset is enabled.

https://user-images.githubusercontent.com/14921654/105219271-903a7f80-5b56-11eb-8215-3daf37fc2465.mp4

liftbag, 2021-01-22:
Looking back, the layer disappears because the first layer of the object is extruded at double height. This behavior is inherited from PrusaSlicer, as described in the issue [3425](https://github.com/prusa3d/PrusaSlicer/issues/3425).
Although I never use the raft, I understand the concerns of those who would not want the first layer of the object to be extruded at a layer height equal to the nozzle diameter.

https://user-images.githubusercontent.com/14921654/105506802-bc770d00-5cca-11eb-9197-688c6c4c72a8.mp4

neophyl, 2021-01-22:
This exact same issue has been opened on the prusa slicer github too https://github.com/prusa3d/PrusaSlicer/issues/5835

supermerill, 2021-01-23:
> Although I never use the raft, I understand the concerns of those who would not want the first layer of the object to be extruded at a layer height equal to the nozzle diameter.

But as it's over a raft, it needs to be a "bridge", ie a nozzle diameter height.
Or it will increase the height of the object artificially.