supermerill/SuperSlicer #900: default acceleration value is not being emitted to g-code if it's the only one that's set; first layer acceleration is similar

VanessaE, 2021-01-20:
### Version

commit b6d3143c868c4713f71c20e64d4cc631c93e8400

### Operating system type + version

Debian Bullseye/XFCE 4.16

### 3D printer brand / version + firmware version (if known)

N/A

### Behavior

Acceleration commands are not being emitted into the g-code if only one field, whichever it may be, is set to a non-zero value.

For example, if I set the "default acceleration" to some legit value, say 1234, and all other values to 0, I would expect to find a single `M204 P1234` command somewhere near the start of the g-code file, and then no others for the rest of the file.  However, not even that one command is emitted, so the file would get printed with whatever random acceleration was last set by some other print.

Similarly, if I were to set "first layer acceleration" to some value, and then everything else to 0, no corresponding `M204` command is emitted into the g-code (and all fields except for default acceleration get greyed-out, although this is likely intentional).

On the other hand, if I set the default acceleration to some value, say 1234, and set first layer acceleration to some other value, say 555, and then set everything else to 0, then the exported g-code will have `M204 P555` near the start, a sprinkling of that and `M204 P1234` throughout the first layer, a final `M204 P1234` at the end of the first layer, and then no further acceleration commands after that.

If I set both default and first layer acceleration to the same value, then I get a single corresponding `M204` command near the start of the code. 

In practice, it's probably a good idea to set both the default and the first layer accelerations to appropriate values, but in all cases I would expect the default acceleration value to *always* be written to the g-code near the start, assuming it isn't set to 0, regardless of what the other fields are set to.

#### Project File (.3MF) where problem occurs
[Retraction 2.3mf.zip](https://github.com/supermerill/SuperSlicer/files/5841004/Retraction.2.3mf.zip)

bakaufman, 2021-02-21:
Which flavor is this? Klipper or Marlin?

VanessaE, 2021-02-21:
I'm using Marlin.

mikes2277, 2021-04-18:
I have the same issue with Klipper g-code flavor except there are no M204 commands anywhere in the gcode.  Tried this on 2.3.56a nightly build, downloaded on 4/17/2021.  2.3.55.5 does not have this problem with Klipper g-code flavor and at glance, places tons of M204's in the gcode.

mikes2277, 2021-04-19:
I figured out what is causing it, but I am not sure if this is by design or not.  The machine limits have the following fields:

Max acceleration X
Max acceleration Y
Max acceleration Z
Max acceleration E
Max acceleration when extruding
Max acceleration when retracting
Max acceleration when traveling

It seems that Super Slicer really only honors the "Max acceleration when extruding"  value and ignores the rest.  You can literally put anything you want in the rest of the fields from what I can tell, and they change nothing in the generated gcode.  But if you put "Max acceleration when extruding" to anything lower than your Print Settings->Speed->Acceleration control values, it will not output M204 commands for anything above the number.  At the very least I would think it should output M204's to put it to the machine limit ceiling instead of no M204 command at all, otherwise you may end up with a slower acceleration because a previous M204 command may be for lower than the ceiling.

supermerill, 2021-04-25:
> It seems that Super Slicer really only honors the "Max acceleration when extruding" 

Depends of the firmware...
Also an option let the slicer output them as limits for the firmware
Some aren't used to write gcode but to estimate the printing time.

mikes2277, 2021-04-26:
@supermerill it might make sense to add warnings in the Print Settings->Speed->Acceleration control (advanced) section to warn you that it will not be including the values in the section based of the settings in the Printer Settings->Machine Limits->Maximum Acceleration or Printer Settings->General->Firmware->G-code flavor setting to help out those who don't realize this is happening or don't understand how to read gcode.

supermerill, 2021-04-27:
Is there a way to transmit it to the firmware?

supermerill, 2021-04-27:
@VanessaE 
A bit late, soi i'm testing it with the latest version.
You have to select in 'How to apply limit' 'also emit limits in gcode' instead of 'disable'

VanessaE, 2021-04-27:
No, that's only supposed to be for the hardware limits, which I definitely do NOT want.  My machines already have their hard limits set the way I want them and saved in EEPROM.

`M204` is *default* acceleration, not the maximum limit, and is supposed to be used on every print to indicate "right now", regardless of the state of the hardware limits window.

supermerill, 2021-04-27:
Oh, understood, you're right.
I have to gather all gcode writing problems right now and decide if it's worth to fix before rewriting.

supermerill, 2021-04-27:
seems fixed in my testing

lrstanley, 2021-08-04:
I am running into what may be a similar issue on v2.3.56.7, however it doesn't relate to anything being set to 0. More-so that it only ever adds one `M204` in the generated gcode, and thus will never go above that one reference.

Attached is generated gcode, based off the below config: [Metric_screw_measuring_device_-_thread_markings.zip](https://github.com/supermerill/SuperSlicer/files/6928627/Metric_screw_measuring_device_-_thread_markings.zip)

What's set in the gcode is 1500

[image]

I don't have machine limits enforced under the printer tab, however they're set to 9000 so even if they were, I wouldn't think those configs would apply.

My configs: [configs.zip](https://github.com/supermerill/SuperSlicer/files/6928658/configs.zip)

Anything I'm missing, or is it the same issue? Different issue?

```
SuperSlicer
Version:   2.3.56.7
Build:     SuperSlicer_2.3.56.7

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19043), 64-bit edition
Total RAM size [MB]: 34,307MB
OpenGL installation
GL version:   4.6.0 NVIDIA 466.11
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce RTX 2060/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

[...]
```

lrstanley, 2021-08-04:
Ah, I'm blind. I now see the comment here: https://github.com/supermerill/SuperSlicer/issues/900#issuecomment-822262779

Changing those configs, even those they aren't supposed to have an impact on hardware limits/change anything in the gcode, do look to. I've adjusted everything to be higher than my actual limits to allow it to correctly write all of the necessary M204 gcode lines.

Thanks, @mikes2277!

@supermerill 
> seems fixed in my testing

Is this a non-issue in HEAD? or are you saying you couldn't replicate in ~2.3.56.7?

supermerill, 2021-08-05:
> Is this a non-issue in HEAD? or are you saying you couldn't replicate in ~2.3.56.7?

Can't replicate in my dev branch at the time. You shouldn't be able to replicate in ~2.3.56.7

lrstanley, 2021-08-05:
> > Is this a non-issue in HEAD? or are you saying you couldn't replicate in ~2.3.56.7?
> 
> Can't replicate in my dev branch at the time. You shouldn't be able to replicate in ~2.3.56.7

I can replicate in 2.3.56.7, as shown in my above post https://github.com/supermerill/SuperSlicer/issues/900#issuecomment-892361392

supermerill, 2021-08-06:
@lrstanley please post your project file used to create your g-code.

lrstanley, 2021-08-06:
It looks like the same issue (outlined [above](https://github.com/supermerill/SuperSlicer/issues/900#issuecomment-822262779)), where the "machine limits" being set to only impact time-estimates, is not being obeyed. Some of my configuration was slightly different to others, however, I believe it's still the same issue.

The part in mention: 
[Metric_screw_measuring_device_-_thread_markings.zip](https://github.com/supermerill/SuperSlicer/files/6947598/Metric_screw_measuring_device_-_thread_markings.zip)

Given there is a configuration option that lets you change if the machine limits are ONLY used for time estimates, or if they are also used for actual machine limits (that change GCODE), it's reason to assume that setting the option to "time-estimates only" should only impact time estimates.

If the limits being put in place (`acceleration when extruding` options and similar) are suppose to always have an impact (which may be the case, the wording in the dropdown is a little confusing), then I think the issue is that this isn't obvious, and there should be some clarifiers on those pages. For example, if you set your acceleration on the print page to a higher value, and have your `acceleration when extruding` lower than those values, maybe there is a warning that says "Acceleration is limited to <accel>mm^3/s due to the following setting: <name>`. This could be similar to what we see on the filament cooling page, where it clarifies what the fan speeds are at specific sections, to clarify things a bit.

Either way, I feel that things could either use some clarification in the UI to ensure it's obvious that there is a cross-impact, or ensure there is an option under machine limits that has no impact on limiting the machine speed (i.e. letting the printer controller firmware do the necessary machine limits).