supermerill/SuperSlicer #905: Overhang function

misiek271, 2021-01-21:
### Version
2,3,55,2

### Operating system type + version
win 10 64bit

### 3D printer brand / version + firmware version (if known)
anycubic +mks robin nano v2 

i have request for overhanging. YOu add value of overhangs. thats great but can YOu also add value that can be setup to start overhangs X mm before actual overhangs(lets say its virtual overhang). as you can se often overhangs starts on curve so printer slow down on curve it's isnt great behavior coz the linear advance ect. so it will be great to have option that overhang shoud be start X mm before Real overhgan. so the printer should have time(distance) to slow down and regulate preasure in nozzle to make the corner. 

on my photo i make line to visualise my thinking.
[image]

supermerill, 2021-01-23:
I don't understand perfectly what the problem is.

the speed slowdown is too late?

misiek271, 2021-01-23:
No i mean that You should add something like "transition lenght" from permiter to overhang. 
lets preasume i print permiters with 
50mm/s then my permiters goes to overhang so printer slowes to 10mm/s But preasure in nozzle is big and ad first mm overhang filament is overextruding. 
So it will be nice to add some settings in mm. that tels that printer must slow down from 50mm/s to 10mm/s before overhangs starts. and i declarate that lenght of slow down should be 10mm (user can type own value coz printers has got differnt acceleration /deacceleration settings)
i attach my shema. lets preasume that its a single extrusion line.
[image]

robthide37, 2021-01-25:
shouldn't pressure advance take care of this ?

misiek271, 2021-01-25:
If You have dd yes but on bowden there is no option to set LA in bowden perfectly like on direct crive coz ptfe tension ect. And this kind of feature shoud give a very good results. i think that it will helps on direct drive too. Coz LA is just prediction of preasure in nozzle there is no sensor i nozzle that tels printer what is actual preasure. So This kind of slowing down shoulld perfectly regulate preasure in nozzle. distanse should be able to typ in by user. And when you set this slowing down function let say 20mm long 70%(14mm)  of this lenght should be slowing down from 50mm/s to 10mm/s and 30%(6mm) shoud be 10mm/s speed to help nozzle stabilized preasue.

robthide37, 2021-01-27:
got you i was more curious if one would interfere with the other but does not seem so . One thing that would be maybe helpfull is a way to set accel much lower agead of time that may be a simple way to achieve the same effect or it may not i have not thought it in and out

supermerill, 2021-03-18:
It should lay on the firmware side, imo.
I'll see if i can do something for it when I'll work again on the pressure model.