supermerill/SuperSlicer #908: Incorrect slicing of model

oabdullin, 2021-01-22:
### Version
2.3.55.2 and earlier like 2.3.55.2

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
it's not related to printer

### Behavior
The slicer cant slice this model 

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`File -> Save` / `Save Project`)_

[Small_boat_for_Minifig.zip](https://github.com/supermerill/SuperSlicer/files/5858538/Small_boat_for_Minifig.zip)

[image]

[image]

Pavulon87, 2021-01-22:
Stl with errors.
[image]
[Small_boat_for_Minifig_v2.zip](https://github.com/supermerill/SuperSlicer/files/5858656/Small_boat_for_Minifig_v2.zip)

oabdullin, 2021-01-22:
@Pavulon87, how did you fixed STL?

liftbag, 2021-01-23:
> @Pavulon87, how did you fixed STL?

You can try with [Netfabb online service](https://service.netfabb.com/login.php).

closed 2021-04-27 by supermerill