supermerill/SuperSlicer #909: Color/extruder selection from Layer slider + button

jimhiggs, 2021-01-23:
### Version
2.3.55

### Operating system type + version
W10

### 3D printer brand / version + firmware version (if known)
Custom build, Diamond mixing extruder

### Behavior
Just got into Superslicer, been using kissliser for years.  Looks amazing.
Prusaslicer has the option to change extruder  at a layer by pressing a plus button beside the layer slider after slicing.
Should this be available in Superslicer as I cant find it?
(I have also installed latest Prusaslicer and set up both with same config. It works as expected in PS).

neophyl, 2021-01-23:
I have just tried both the stable 2.2.54 and 2.3.55.2 and both have the colour change/pause/custom g code option on my install.

You do have gcode entered in to the printer settings>custom gcode sections for those commands dont you ?

If you would like to save a project that is suffering the problem (Project >save as) and then take the resulting .3mf file and zip it in to an archive and attach it here we can take a look.  A project file contains all the settings you are using, including your printer settings and custom gcode so we have a chance of figuring it out.  Without a project file its pure guess work.

btw the 2.3.55 releases are built upon the Prusa Slicer early ALPHA builds so contain lots and lots of bugs that were in the alphas as well as issues from the merge.  Expect crashes and other issues when using them.

jimhiggs, 2021-01-23:
Thanks for the update. I downloaded both 2.2.54 and only switched to 2.3.55.2 to see if the color option was in that version as it is was not showing for me in 2.2.54. I have switched back.
I found my issue, I had sequential printing turned on. For some reason I never saw the little icon with the warning up until now. 
Thanks for the help will close this now.

closed 2021-01-23 by jimhiggs