supermerill/SuperSlicer #911: Retract- Extrude values incorrect at start of print

creek52, 2021-01-23:
2.3.55
_Version of SupserSlicer used goes here, Use `About->About SuperSlicer`_

Windows 10 Home. Build 19042.746
_What OS are you using (and version)_

Delta, Duet Wifi, Firmware: RepRapFirmware for Duet 2 WiFi/Ethernet 3.1.1 (2020-05-19b2)
Duet WiFi Server Version: 1.23 
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* At beginning of first layer there is a retract of the filament, then a few lines later there is an extrude filament to replace the retract. The problem is the the extrude  is not the same value as the retract to replace was was retracted. I have to edit gcode file to correct the lack of filament at the start. Part of raft is missing until it extrudes enough to fill from retraction.
*  SuperSlicer
* ; custom gcode end: start_filament_gcode
;LAYER_CHANGE
;Z:0.25
;HEIGHT:0.25
G1 Z0.150000 F7800.000
G1 E-3.60000 F2400.00000   <----this is the retract value -3.6
G1 Z0.850000 F7800.000
G1 X-0.065 Y-34.769
G1 Z0.150000
G1 E0.00000 F2400.00000    <----this should extrude the opposite value of 3.6, E3.6 F2400.00000
;TYPE:Skirt

PrusaSlicer 2.3.0 +win64
;LAYER_CHANGE
;Z:0.25
;HEIGHT:0.25
G1 Z0.150 F7800.000
G1 E-3.60000 F2400.000     <----same retract
G92 E0
G1 Z0.150 F7800.000
G1 X-33.436 Y1.680
G1 Z0.150
G1 E3.60000 F3000.000       <-----correct extrude
;TYPE:Skirt

  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*SuperSlicer*__ preview are preferred_

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`File -> Save` / `Save Project`)_

tirasec, 2021-01-24:
Looks correct ...
Susi Retract -3.6mm 
G1 E-3.60000 F2400.00000  (extruder position is -3.60)
then extrude back to E0
G1 E0.00000 F2400.00000 

PrusaSlicer Retract -3.6mm 
G1 E-3.60000 F2400.000
then set extruder position to 0
G92 E0
Then extrude 3.6mm
G1 E3.60000 F3000.000

supermerill, 2021-03-18:
@creek52 You don't gave your project file. Are-you in relative mode or not?