supermerill/SuperSlicer #912: Need function One loop perimeters two first layer only ?

Witawat, 2021-01-24:
Need function One loop perimeters two first layer only ?

i try function One loop perimeters is good. but not work use
all layer this seam and loop perimeters not ok perimeters layout
not good or can limit One loop perimeters in layer ? or can use
seam and loop seam in one position ?

thankyou

supermerill, 2021-01-24:
![image](https://user-images.githubusercontent.com/6536403/105636926-ea935300-5e6a-11eb-87a8-3b834509c486.p

1) add a layer range modifier
2) set it to the 2 twxo first layers (change the stop at height box)
3) add settings -> perimeter to it
4) select periemeter loop
5) activate it

LazaroFilm, 2021-03-19:
I also use single perimeter on first layer on my prints and would love to see a single click option to make this happen. It's really getting old to have to dial this every time It takes 6 to 10 actions (clicks and keystrokes) per object to get that. I would love to see a single box check similar to the top layer option or a Layer Range Setting preset that you could save and apply to all your parts. Thanks for your consideration.

supermerill, 2021-03-20:
@LazaroFilm 
yes, it's discussed in #412 and #750

closed 2021-06-04 by supermerill