supermerill/SuperSlicer #913: unwanted layer temp changes after printing temp tower

gregorve, 2021-01-24:
### Version
2.2.53

### Operating system type + version
WIN 7 Pro

### 3D printer brand / version + firmware version (if known)
Ender 3 Pro, Octoprint, Klipper

### Behavior
after printing the temp tower (from calibration menu) the temperature changes on layer change remain in the gcode when other stl files are sliced after that.

Restart of SuperSlicer fixes the issue.

supermerill, 2021-01-24:
These temp changes are not linked to the part.

I'm wondering what is the best: adding a pop-up for every time you remove everything and have some custom layer gcode?
Or just an other pop-up with more option for when you click on "delete all" ?

cbusillo, 2021-01-24:
Ha, this one got me yesterday.  It wasn't obvious to me where the temp was being set at.  I switched back to PrusaSlicer for a few prints.  Restarting fixed it though.  Keep up the great work!

azaz44, 2021-03-13:
One improvement to do would be to not disable "New Project" in such case. If you remove everything by hand, then you can't do "New Project". And actually this would clear the g-codes.

supermerill, 2021-03-18:
#949