supermerill/SuperSlicer #914: [2.3.55.2] Bridge Perimeters Cause Infill Split Above

CCS86, 2021-01-24:
2.3.55.2 / Win 10 x64

It seems as though bridge perimeters are not being considered "bridges", and are causing the infill area above to split, with a non-100% above-bridge flow ratio:

[image]

[image]

[Bridge Test.3mf.txt](https://github.com/supermerill/SuperSlicer/files/5862843/Bridge.Test.3mf.txt)

IMO, "bridges" and "bridging perimeters" should be considered the same for this purpose and not split.

qwewer0, 2021-02-05:
The issue is present on 2.3.55.4.

To reproduce the issue, use `over_bridge_flow_ratio != 100%`,
and for better visualization use `only_one_perimeter_top = 1` with `perimeters > 1`

CCS86, 2021-02-05:
> The issue is present on 2.3.55.4.
> 
> To reproduce the issue, use `over_bridge_flow_ratio != 100%`,
> and for better visualization use `only_one_perimeter_top = 1` with `perimeters > 1`

2.3.55.4?

I don't see that in releases.

qwewer0, 2021-02-05:
@CCS86 You can find it in the [Actions](https://github.com/supermerill/SuperSlicer/actions) tab

supermerill, 2021-03-18:
That's because of the over_bridge_flow_ratio. That is used to compensate a bit the drop from the bridge area.
You can set it to 100% to remove this split when it's not needed.

supermerill, 2021-03-18:
mmm. Maybe the problem is that the over_bridge_flow_ratio should also be used over the overhangs ?

qwewer0, 2021-03-18:
> mmm. Maybe the problem is that the over_bridge_flow_ratio should also be used over the overhangs ?

@supermerill That might be a good idea, but I think this issues main point it that the over_bridge_flow_ratio line isn't printed continuously with the other lines.

CCS86, 2021-03-18:
> mmm. Maybe the problem is that the over_bridge_flow_ratio should also be used over the overhangs ?

I would tend to say yes to this. But, I only use "overhang perimeters" as "bridge perimeters", with a 100% unsupported requirement. 

Maybe "bridge perimeters" and "overhang perimeters" need to be split into discrete categories? Then, bridges are bridges, whether perimeter or internal, and can be combined for these flow overrides.

Also, the "above bridge" flow override seems to do nothing (besides splitting the infill). Flow rate is not increased at all. I can start a new issue if you would like.

supermerill, 2021-03-19:
If I remember, it doesn't increase flow rate but creates more overlap, so it extrudes more plastic in the same area.

qwewer0, 2021-06-04:
I think this issue is still present on the latest release.

over_bridge_flow_ratio: 99%
[image]

over_bridge_flow_ratio: 100%
[image]

supermerill, 2021-06-04:
there is already issue #234 to remind me of it.

supermerill, 2021-06-04:
> I think this issue is still present on the latest release.

it's something I wanted, that's the whole point of the feature.
This can only go away when modifiers will stop to create split in perimeters & infill.

qwewer0, 2021-06-04:
> it's something I wanted, that's the whole point of the feature. Use 3 top layers or no over-bridge flow (100%) to don't see it.
> This can only go away when modifiers will stop to create split in perimeters & infill.

Yes, but as we now know why this happens, then could we see https://github.com/supermerill/SuperSlicer/issues/914 as a feature request?

supermerill, 2021-06-04:
if you want.
there is also #1092 that is the same thing but for perimeters.

qwewer0, 2021-06-04:
> if you want.
> there is also #1092 that is the same thing but for perimeters.

Thank you, very kind of you.