supermerill/SuperSlicer #916: "Monotonic (filled)" infill on bottom layer is not monotonic

arkeet, 2021-01-25:
### Version
2.3.55.2

### Operating system type + version
Windows 10 2004 amd64

### 3D printer brand / version + firmware version (if known)
Voron 0 + Klipper (shouldn't matter)

### Behavior
When selecting "Monotonic (filled)" for bottom infill, the resulting infill is not monotonic (it looks the same as "rectilinear" but filled).
* "Monotonic (filled)" works correctly for top infill.
* "Monotonic" (not filled) works correctly on bottom infill.

Example first layer, chronology view:
[image]

#### Project File (.3MF) where problem occurs
[end_plate.3mf.zip](https://github.com/supermerill/SuperSlicer/files/5864226/end_plate.3mf.zip)

qwewer0, 2021-01-29:
Did you draw the conclusion from the chronology view?

<details>
<summary>Monotonic (filled)</summary>

<details>
<summary>Top</summary>

[image]

</details>

<details>
<summary>Bottom</summary>

[image]

</details>

</details>

<details>
<summary>Monotonic (not filled)</summary>

<details>
<summary>Top</summary>

[image]

</details>

<details>
<summary>Bottom</summary>

[image]

</details>

</details>

arkeet, 2021-02-02:
I first noticed it while actually printing. Chronology view just gives a way to visualize it (redder parts are printed before greener parts, so there should not be a green part between two red parts in the infill, or vice versa)

qwewer0, 2021-02-02:
> I first noticed it while actually printing. Chronology view just gives a way to visualize it (redder parts are printed before greener parts, so there should not be a green part between two red parts in the infill, or vice versa)

Ok, that makes sense. I  will test it to see it with my own eyes too.

qwewer0, 2021-02-02:
@arkeet
https://imgur.com/gallery/cFzA2eQ

1. **Monotonic (not filled)** infill has a hiccup at **21sec left** (prints in between already printed area) in the video, so it has problems too.
2. **Monotonic (filled)** infill prints left-right all over the place, so it doesn't seem to be monotonic.
3. PrusaSlicer Monotonic infill seems to work.

qwewer0, 2021-02-03:
@arkeet Could you update the original description and title, if you are able to reproduce the problem with **Monotonic (not filled)** in the video above?

arkeet, 2021-02-03:
Since you already observed it could you share your project file?

qwewer0, 2021-02-03:
@arkeet Sure, but you might want to try it with your own settings too.
[Monotonic-3mf.zip](https://github.com/supermerill/SuperSlicer/files/5920789/Monotonic-3mf.zip)
[Monotonic-test-slt.zip](https://github.com/supermerill/SuperSlicer/files/5920805/Monotonic-test-slt.zip)

qwewer0, 2021-02-04:
The issue is present on 2.3.55.4

arkeet, 2021-02-23:
This appears to be fixed in 2.3.55.5. @qwewer0 can you confirm?

qwewer0, 2021-02-23:
@arkeet
On 2.3.55.5, Monotonic for bottom layer doesn't work.

Monotonic (**not** filled) bottom layer:
[image]

supermerill, 2021-03-18:
using @arkeet model and using monotonic inside it, I can't spot a not-monotonic section.

Can someone share a buggy project on 2.3.5.55 ?

qwewer0, 2021-03-18:
> using @arkeet model and using monotonic inside it, I can't spot a not-monotonic section.
> 
> Can someone share a buggy project on 2.3.5.55 ?
Re tested it on 2.3.55.5 and it seems to be working for both top and bottom with filled and normal monotonic infill.