supermerill/SuperSlicer #917: Fusio 360 link to SuSl

Gertsch, 2021-01-25:
### Version
2.3.55.2

### Operating system type + version
WIN10

### 3D printer brand / version + firmware version (if known)
Ender-3, M4 extruder, Klipper

### Behavior
Link from Fusio360 to SuSl is not working.
When I go to "Make" and try send to Ss it doesn't do anything unfortunately.

supermerill, 2021-03-18:
?

I don't use fusion360, and i don't understand what the "make" button do.

Gertsch, 2021-03-18:
sorry, haven't noticed your response!

The "make" button is a link between fusion and a slicer of your choice. It automatically exports the cad to the slicer.

supermerill, 2021-03-19:
Can you know what is the command line it outputs?

something like `SuperSlicer.exe mystl.stl` ?

unkn0wndude, 2021-03-19:
I have no issues with this. What I do is "Save as STL" in F360 and then check "Send to 3D Print Utility", Print utility: "Custom", Application: "superslicer" (path to SS exe). It opens the new SS windows with stl imported.

closed 2021-06-14 by supermerill