supermerill/SuperSlicer #920: [Feature request] Show differences when re-slicing the same model with changed settings

JohanAR, 2021-01-25:
Mockup:

[image]

I think this would make it easier to compare different print settings that one is considering

supermerill, 2021-03-19:
Why not. I'll look at it when I have to modify something in this gui window.

JohanAR, 2021-03-19:
Qwewer brought this up on the discord:

> #920 would be really nice addition to SS, but two thing that needs to be counted in is the Background processing and the Layout Option.
> If Background processing is enabled and
> 1. If Layout Option is either the "Old regular..." or the "New layout..." then the print time differences should only update when we are in the Plater windows/tab, because if the time is updated continuously as the part is sliced, then after multiple settings changes the time difference will be different to what we expect if we would slice manually.
> 2. If Layout Option is "Settings in non-modal window", then it should either update as a setting is changed or only update when returning to the Plater, but I think in this care the first is the better option.

supermerill, 2021-03-19:
I'm more leaning towards
* If Background processing: compare with "saved settings run" (if it doesn't exist, then show nothing)
* if not, compare with the previous slice