supermerill/SuperSlicer #921: Feature Request: Tool Pathing and Extrusion Width clarification

Bobbiebooties, 2021-01-26:
### Version
2.3.55.2

### Operating system type + version
Not Important

### 3D printer brand / version + firmware version (if known)
Not important

### Behavior
Problem:
      Extruder pathing is dependent on both nozzle diameter, and extrusion width.
Reproduce:
      Add a shape, any shape. Modify extrusion width, and pay attention to the extruder pathing.
Expected results:
      Tool pathing should be hard linked to the nozzle diameter, not extrusion width.
Actual results:
      Tool pathing is hard linked to the extrusion width and nozzle diameter.

The main issue is that there is no way to adjust the extrusion width without affecting tool pathing.
This can be somewhat modified by changing the flow settings, but that only works for some feature types (Bridges, Above the bridges, Top Fill, and First layer).

Adding more flow modifiers wouldn't fix the core issue of how pathing and extrusion are obfuscated from the end user.

For example:
Using a 1.0mm nozzle, I want the tool paths to be 1.0mm apart.
The default extrusion width would be 1.0mm (100%), meaning the extruded plastic should just barely touch.
If I wanted the extruded lines to have more adhesion, I could simply increase the extrusion width without affecting the tool paths.

A separate multiplier could then be applied to the nozzle diameter to modify the tool path spacing.

CCS86, 2021-01-26:
Check out my issue here. We launched deep into that discussion:

https://github.com/supermerill/SuperSlicer/issues/592

Bobbiebooties, 2021-01-26:
I didn't check that issue because it didn't mention extrusion width in the title.
I guess this has already been discussed, but at least I've learned that overlap is likely the reason behind my poor quality prints.

CCS86, 2021-01-26:
I don't think overlap is causing your poor quality prints. The amount of overlap is mathematically accurate:

https://manual.slic3r.org/advanced/flow-math

You can even disable that if you want with this setting:

[image]

IMO, this should be handled internally. Meaning that the user specifies the line spacing (center-to-center), and the line width is calculated internally, based on the commanded spacing and overlap %. This would make it much easier to model in increments of nozzle width (ex. 0.4 x 10), command a 0.4 line spacing, and get the expected number of lines, with no gap.

Bobbiebooties, 2021-01-26:
Overlap would cause print problems if you tune your e-steps to account for a different flow model.
To put this into perspective, I was getting glass smooth finishes with Cura, but there was always some overextrusion since I've changed slicers.
I don't want to have to recalibrate e-steps for every slicer, because there are still some times I have to use Cura even though I hate it.

I started another test print with 0% overlap after my previous post, and I'll post an update when it's done.

Bobbiebooties, 2021-01-26:
Yup. Overlap was the cause.
107% extrusion width from #592 and some overlap tuning will get close to what I want.

The issue still stands though. There is no clear and concise method to define a tool path and extrusion width.

CCS86, 2021-01-27:
My guess is that your 1mm nozzle just behaves a little differently to a 0.4mm nozzle, in regards to the appropriate amount of overlap. Once you find the overlap value that works well with your extruder calibration, it shouldn't need to change much. maybe a little bit depending on layer height.

I think my suggestion for a SS update would make us both happy though. Giving the user direct control of path spacing and line width, then using background math (with a user setting) to tune flow rate for filling the non-rectangular voids of the line intersections. Basically, making it a calculated "over-flow" instead of a variable overlap.

Bobbiebooties, 2021-01-27:
I completely agree with what you're saying. #592 and the other related issue were the route I took to remedy the <1% over-extrusion on my top surface.

My solution was to increase the top layer extrusion width to 225%, which increases the spacing between extruder paths, and then decrease the top surface infill flow to 50%, with a perimeter-infill overlap of 0%.
There are defined valleys between extruded lines, but at least there are absolutely no high spots from 0.25% over-extrusion defects in consecutive solid infill features.

supermerill, 2021-03-18:
> Using a 1.0mm nozzle, I want the tool paths to be 1.0mm apart.

For the perimeters, if you put 0% in all "perimeter overlap" settings , and 1mm in all perimeter extrusion width it should do that.
(but not for solid infill)

Note that you'll get perimeter separation with this (as they won't touch each other anymore)

CCS86, 2021-03-18:
I make a lot of functional parts, so I design a lot of part wall thicknesses as multiples of nozzle width (ie: 0.4 * 12).

It would be great if SS could allow the user to define the "effective line width" directly; and using the overlap % specified to calculate the [path spacing] and [actual line width] internally.

Here is a drawing to illustrate:

[image]

We have a part with a 1.2mm wall thickness, and we want 3 walls, so we want a 0.4mm effective line width.

Using layer height and overlap factor, we can determine the actual line width (0.43) an the line spacing (0.385)

When we add another wall, the values change slightly, but the concept is the same.

Does this seem practical?

CCS86, 2021-03-18:
Here is is projected out to 16 walls, just to make sure the actual line width didn't drift too far. Looks okay to me:

[image]

supermerill, 2021-03-19:
what i can do is to allow to set the width of the extrusion or the spacing of it. 

spacing = extrusion_width - layer_height * (1 - PI/4)

if you want a spacing of 0.385 with 100% overlap ( no change) and 0.2 layer height, you need to set an extrusion_width  of 0.428

CCS86, 2021-03-22:
> What i can do in the future is to allow to set the width of the extrusion or the spacing of it, as I clearly see that this confuse a lot of users.
> 
> spacing = extrusion_width - layer_height * (1 - PI/4) * overlap_percentage
> 
> So if you want a spacing of 0.385 with 100% overlap ( no change) and 0.2 layer height, you need to set an extrusion_width of 0.428
> 
> if you want an extrusion_width of 0.43 and spacing of 0.385 then you need an overlap of 104.6%, but it's not possible to go over 100% because this leads to over extrusion, it's like asking for more plastic than the volume can accept.

If you look in my other post, I wrote about the 0.045mm overlap being a guess. 0.043mm is the correct overlap for a 0.2mm layer and 100% fill (with that model).

The problem I see with defining path spacing that way, can be seen in the sketches I made. Depending on the amount of lines you are trying to fit between two outer surfaces, and keeping a fixed amount of overlap, the line spacing needs to change very slightly. This is because you have two outward facing lines without any overlap, and depending on how many total lines there are, these outer lines contribute a different % to the total width.

Any ideas on the best way to resolve this?

supermerill, 2021-03-22:
You can ask for an external perimeter "width&spacing" and internal perimeter spacing.
for exemple, if you ask for a 0.5mm "width&spacing" and 0.5 mm spacing, with3 perimeters, the 2 outer loop& 4 inner loop will take exactly 3mm (if you have a 3mm thick wall). The slicer compute itself the correct width for the internal & external perimeter (0.543 with for internal, 0.5215 for external with 0.2 width and 100% overlap)

CCS86, 2021-03-22:
That sounds perfect!

CCS86, 2021-03-22:
Slightly off topic: why isn't there an overlap parameter for solid infill? 

Is it just a fixed amount of overlap, or is the perimeter parameter applied?

[image]

supermerill, 2021-03-22:
> why isn't there an overlap parameter for solid infill?

I don't see the need. I added these settings for perimeters because perimeters are visible and this setting may help to resolve issues like benchy line. 
Decreasing the overlap on solid infill will result on underextrusion, and a top layer that is a bit sagging (even more that currently).

CCS86, 2021-03-23:
Cool, that makes sense.

supermerill, 2021-06-04:
Current "spacing settings should resolve this issue unless I misread it.

closed 2021-06-04 by supermerill