supermerill/SuperSlicer #923: After restarting SuperSliser, the code page is lost.

user32rtu, 2021-01-28:
### Version
2.3.55

### Operating system type + version
Windows 10 Pro 64 bits 20H2

### 3D printer brand / version + firmware version (if known)
Sapphire Pro

### Behavior
I write comments in Start G-code in Russian and I see Russian text. After restarting SuperSliser, I see that where I wrote in Russian, the code page has now changed and the text becomes unreadable.
[image]
[image]

neophyl, 2021-01-28:
Having recently gone through the pain of adding non english language support to one of our products at work this doesnt surprise me.  The configuration file is probably only stored/coded to allow a normal latin character encoding once saved.  When read back what is stored isnt quite what you wrote.

supermerill, 2021-03-18:
You can try to activate the utf-8 beta mode in windows.

user32rtu, 2021-03-20:
"You can try to activate the utf-8 beta mode in windows."

Thanks for the advice. It works as it should.