supermerill/SuperSlicer #924: Convert round holes to polyholes option does not work

user32rtu, 2021-01-29:
### Version
2.3.55

### Operating system type + version
Windows 10 Pro 64 bits 20H2

### 3D printer brand / version + firmware version (if known)
Sapphire Pro
### Behavior
When I enable the "Convert round holes to polyholes" option, when I view the G code, I do not see any changes in the shape of the holes. The small hole in the screenshot consists of 32 segments when the setting is on and off.

#### Project File (.3MF) where problem occurs
[image]
[image]
[image]
[image]

[ScaleHole_v2.zip](https://github.com/supermerill/SuperSlicer/files/5891473/ScaleHole_v2.zip)

qwewer0, 2021-01-29:
There is something weird going on. (3rd smallest hole)
[PolyHoleTest_3mf.zip](https://github.com/supermerill/SuperSlicer/files/5893232/PolyHoleTest_3mf.zip)
Top of the part:
[image]
Inside of the part:
[image]

user32rtu, 2021-01-29:
The error still exists.
[PolyHoleTest_3mf.zip](https://github.com/supermerill/SuperSlicer/files/5894105/PolyHoleTest_3mf.zip)
[image]

qwewer0, 2021-02-04:
The issue is still present on 2.3.55.4

supermerill, 2021-03-21:
hole polygon wasn't regular. I'm now a bit more lenient in the detection.

qwewer0, 2021-03-22:
@supermerill It is still not really working (latest nightly_win64)

[Polyhole.zip](https://github.com/supermerill/SuperSlicer/files/6182122/Polyhole.zip)
https://imgur.com/gallery/DRw1j72

supermerill, 2021-03-22:
@qwewer0 where is the problem?

qwewer0, 2021-03-22:
> https://imgur.com/gallery/DRw1j72

@supermerill The third smallest hole is't converted on every layer

supermerill, 2021-03-22:
ok, i'll see

supermerill, 2021-05-02:
I'll add a user setting to allow for not-extremely-round holes.

closed 2021-06-04 by supermerill