supermerill/SuperSlicer #925: Bad Top-Fill Issue compared to PrusaSlicer

Photogad, 2021-01-29:
### Version
Latest

### Operating system type + version
Win64

### Behavior

The top fill on my object, when using SuperSlicer, puts weird infill on the solid layer instead of a smooth surface, below some raised text on the object. In PrusaSlicer, it does not do this problem, and has a nice smooth surface below the raised text. Both slicers are using the same profile settings that I can see. I tried matching them the best I can, but SuperSlicer have some options PrusaSlicer don't.

Please help!

See photos below.

[image]
[image]
[image]

#### Project File (.3MF) where problem occurs

[shannysugars.zip](https://github.com/supermerill/SuperSlicer/files/5895313/shannysugars.zip)

Photogad, 2021-01-29:
@supermerill 

Actually I found the cause of the issue:
This is happening when both "Enforce 100% Fill Volume" and "Only One Perimeter On Top Surfaces" is enabled.

if I disable both, it fixes the issue.

supermerill, 2021-03-18:
> This is happening when both "Enforce 100% Fill Volume" and "Only One Perimeter On Top Surfaces" is enabled.

You only have to disable "Only One Perimeter On Top Surfaces"

See #733 #945

closed 2021-06-04 by supermerill