supermerill/SuperSlicer #928: Bug: Supporting Dense Layer > Automatic for small layers option does not fall back for larger layers

Hello1024, 2021-01-30:
### Version
Built from master

### Behavior
The "Automatic for small layers" option should fall back to "Anchored" in case a layer is large.  Otherwise we end up with bridges which have no support under the edges and corners where they don't meet a lower infill layer.  The "Anchored" option expands the bridged region by a safety margin.  This still leaves some unsupported edges of the bridge area, but as long as the safety margin is bigger than the infill spacing, the unsupported regions won't be exposed on the finished product.

I will send a PR with a fix shortly.

#### Project File (.3MF) where problem occurs
[3DBenchy.zip](https://github.com/supermerill/SuperSlicer/files/5897990/3DBenchy.zip)
 (between layer 102 & 103)

qwewer0, 2021-02-06:
@Hello1024 Could you post an image of the problematic area? Can't seem to figure out what is wrong at layer 102-103.
The only thing I found that might be wrong is at layer 100, that it doesn't have a coarse bridge layer like at layer 51 and 75, but I'm not sure about that.

supermerill, 2021-02-22:
I'm going to create a new entry for that, as it's not exactly the same behavior.

commit d25f1bff (2021-02-22): Automatic dense infill in large areas now falls back to the Enlarge algorithm rather than being omitted altogether (Fixes #928)
closed 2021-02-22 by supermerill