supermerill/SuperSlicer #929: Ironing in combination with autospeed breaks speed optimizer.

fl0r1s, 2021-01-30:
### Version
2.3.55.2

### Operating system type + version
Windows 10, 10.0.19041 N/A Build 19041 

### 3D printer brand / version + firmware version (if known)
CR10s / klipper

### Behavior
* _Describe the problem_
This bug is related to AutoSpeed. I am getting super slow print times with "ironing" enabled. This is also reflected in the generated gcode.
* _Steps needed to reproduce the problem_
Set all speeds to 0 under speed, enable AutoSpeed with a particular volumetric flow. Next, enable ironing (under infill).
* _Expected Results_
The estimated print time becomes slightly larger due to the ironing operation, print speeds are unaltered.
* _Actual Results_
The estimated print time is 10x the time without ironing, e.g. from 3h30 to 2d4h.

supermerill, 2021-03-18:
fixed by prusa in january, it should be in the latest release. Can you test?

clinta, 2021-03-30:
I'm also seeing this in version 2.3.55.5. Print speeds are very slow if Infill -> Infill -> Top is set to Ironing or if Enable ironing post-process is enabled.

fl0r1s, 2021-04-04:
This is indeed fixed in 2.3.55.5, 

Default ironing speed is 15mm/s, so its normal that your prints will take alot longer if you have large top surface areas.
Thanks!

closed 2021-04-04 by fl0r1s

clinta, 2021-04-05:
But this is affecting much more than the surface areas. See these two gcode previews. The entire object is affected, even infill. This is on 2.3.55.5.

With rectilinear top layer
[image]

With ironing top layer
[image]

fl0r1s, 2021-04-05:
Have tested and was functioning as expected: 

[image]

Some questions pop to mind: 
- Are you running 2.3.55.5?
- Did you disable minimum layer time (filament > cooling)?

clinta, 2021-04-05:
I am running 2.3.55.5. And the only difference between the two screenshots I posted is the top layer setting, so minimum layer time is the same on both, set to 5 seconds. I went ahead and turned off minimum layer time to test, and it does not change the speed, still painfully slow, 2.6mm/s when using ironing top infill.

fl0r1s, 2021-04-09:
Thats weird, Can you please post your config file? Under windows its located at `%APPDATA%/Roaming/SuperSlicer-alpha/printer/<your_profile>` and  `%APPDATA%/Roaming/SuperSlicer-alpha/filament/<your_profile>` 

See below a slice of the same object using my settings: 
[image]

clinta, 2021-04-09:
Configs attached.

[configs.zip](https://github.com/supermerill/SuperSlicer/files/6287523/configs.zip)

fl0r1s, 2021-04-09:
Your config looks fine, Maybe backup configs & reinstall? One last thing is clearing the cache, but other than that I don't think I can be of any more assistance, sorry.

clinta, 2021-04-09:
I'll test with fresh configs, and test with prusaslicer too.