supermerill/SuperSlicer #932: Add multithreading please!

ZhenyaKa, 2021-01-31:
Some models sliced very slow. SS uses one thread only.

JohanAR, 2021-01-31:
When I start SuperSlicer, open a model and slice it has 15 threads and uses all 8 cores to 100%. Which version are you using?

ZhenyaKa, 2021-01-31:
> When I start SuperSlicer, open a model and slice it has 15 threads and uses all 8 cores to 100%. Which version are you using?

2.3.55.2

ZhenyaKa, 2021-01-31:
Just one of six :(
[image]

ZhenyaKa, 2021-01-31:
PrusaSlicer uses one thread too :(

JohanAR, 2021-01-31:
It was a while since I used that version of Windows, but I think Task Manager only shows processes and not threads.

chrome.exe shows up multiple time because each tab is a separate process.

ZhenyaKa, 2021-01-31:
> It was a while since I used that version of Windows, but I think Task Manager only shows processes and not threads.
> 
> chrome.exe shows up multiple time because each tab is a separate process.

I have 3d scanner software and it loads my CPU fully (100%)

JohanAR, 2021-01-31:
I just rebooted to Windows 10 and installed SuperSlicer there, it definitely uses all cores for me on both Ubuntu and Windows..

[image]

ZhenyaKa, 2021-01-31:
I run it at Win7-64x

JohanAR, 2021-01-31:
That could be the reason why we're getting different results. I don't know it's normal that multithreading works differently in Win7 and Win10, but at least we know that this is more likely a bug affecting you, and not a missing feature in SS.

Unfortunately I'm no Windows expert, but perhaps if you post your complete hardware info and Windows build number someone else might be able to assist.

ZhenyaKa, 2021-01-31:
> That could be the reason why we're getting different results. I don't know it's normal that multithreading works differently in Win7 and Win10, but at least we know that this is more likely a bug [...]

how to do that?

JohanAR, 2021-01-31:
I mean just stuff like which CPU you have, how much RAM, if it's a laptop then its model name/number, etc.. To get the build number in Win7 I found [this guide](https://ccm.net/faq/15023-windows-7-check-the-current-build-version) but I haven't tried it myself. It was too long ago since I used Win7 to remember. 

Unless I'm mistaken, Microsoft stopped releasing updates for Win7 about a year ago. Did you run Windows Update at that time so you got all the existing updates? I'm asking because my wife had to replace her motherboard and CPU, and then Microsoft locked her out of Windows Update and told her that her new CPU required Windows 10 even though everything seemed to work just fine

Can't guarantee that it solves your problem, but it often helps people who look at bug reports if they can compare their own system with yours :)

ZhenyaKa, 2021-01-31:
> build

AMD FX-6300, 32Gb RAM
Win7 64x Build7601

JohanAR, 2021-01-31:
I asked if anyone else is using SS on Win7 on the discord server, but no answer yet.

If you get the same behaviour in PrusaSlicer maybe it would make sense to report an issue to them, since SuperSlicer isn't the origin of the problem

CCS86, 2021-02-01:
Multi-threading works fine for me in Win10 x64 / 2.3.55.2

Although, even with the same model, I get different CPU loading during different runs of the slicer:

[image]

[image]

[image]

ZhenyaKa, 2021-02-01:
Very strange :(
Maybe Win7 works bad

neophyl, 2021-02-02:
Or quite possibly a difference between Intel and AMD processor architecture.  Every instance I've seen on here and the Prusa forum/github where someone posts that it is multi core processing wonderfully is with an Intel cpu.
Just wondering if those with AMD chips are the ones having issues.

CCS86, 2021-02-02:
Look above. I'm on AMD.

neophyl, 2021-02-02:
Yes I saw, but didn't you also say that you get different results even with the same model when slicing ?  I'm just wondering if there is something in the way that AMD does multithreading or that the OS interacts with AMD chip(s) that could be a problem with the way that things are implemented.

I also didn't say that all AMD setups are having problems, I said every instance that I've seen which reported zero issues also happened to be on a Intel CPU, (which is admittedly a limited subset as the reports are not widespread).  

There are so many things it could be that way more data is required.  It could be OS, it could be certain chipsets, it could be some weird environment setting in an OS.  Or any of a 1000 things.  The issue was started because they wanted Susie to be multithreading, it is for most people so its probably not a slicer bug.  That doesn't help the original poster (or those others) that its only using a single core on.  More data means the greater chance of finding a common denominator.

JohanAR, 2021-02-02:
OP's AMD is a lot older than yours though, but just looking at its specs it appears to have all the common SIMD extensions..

OP could go to Help -> System Info in SuperSlicer, and see if the last line says "Eigen vectorization supported: SSE, SSE2".. I think this is only one dependency (libeigen) out of many that uses SIMD though

neophyl, 2021-02-02:
Well I have now seen a report of someone with an intel processor based machine that is also only using a single core.  This is with PS, posted on the Prusa forum.  i7 on windows 10.  So that would rule out operating system and cpu from the common factors.

qwewer0, 2021-02-02:
Multithreading works for me.
SuperSlicer 2.3.55.2
Windows 10, 10.0.19042 Build 19042
[image]

supermerill, 2021-03-18:
note: I didn't read everything.

Gcode generation is currently mono-thread, and i have a big project to rewrite most of it to multi-thread it.
Also, the multi-threading is done on the layer level. if you have only 4 layers, you won't see more than 4 threads working at the same time.

closed 2022-04-01 by supermerill