supermerill/SuperSlicer #935: Crash - Sparse Monotonic 100% infill

S-ed, 2021-02-02:
### Version
2.3.55.2

### Operating system type + version
Windows 10 20H2

### 3D printer brand / version + firmware version (if known)
Ender 3 Clone (marlin)

### Behavior
- Open Print Settings -> Infill
- Set Sparse to -> Monotonic
- Set Infill % to 100%
Instant crash

#### Project File (.3MF) where problem occurs
Can crash on an empty project

qwewer0, 2021-02-02:
I too was able to reproduce the issue. (2.3.55.2)

qwewer0, 2021-02-04:
On 2.3.55.4 the slicer is no longer crashes.

uepsie, 2021-02-07:
craches on 2.4.55.4 here too when changing Top layer infill to Monotonic (filled) - non filled and bottom work fine
[image]

-------

[image]

qwewer0, 2021-02-07:
@uepsie On 2.3.55.4? It doesn't crashes for me on 2.3.55.4.
And the crash that I experienced on 2.3.55.2 isn't an error message, just the slicer exits without any warning.

As a side note the error from your images are present on 2.3.55.4, but I didn't got it from the monotonic infill.

uepsie, 2021-02-07:
It does crash for me this way as soon as I chose monotonic (filled) for top layer.

supermerill, 2021-03-18:
can't reproduce on 2.3.55.5
Should be fixed

closed 2021-04-27 by supermerill