supermerill/SuperSlicer #937: Feature Request: configurable fan speed for supports

tecnopolis-ve, 2021-02-02:
Hello

I would like to suggest a simple but useful setting: the ability to set a custom fan speed for support, my default fan speed for PLA is 65% (except for perimeters and bridges) and sometimes this factor makes the support structures really hard to remove, I think that 100% fan speed for supports would be a fix.

Thanks!

user32rtu, 2021-02-18:
I think it is better to cool the layer that falls on the supports. Then it doesn't stick to them much.

tecnopolis-ve, 2021-02-18:
@user32rtu the layer that lays on the support/support interface is cooling at 100% fan speed, the suggestion that I propose is about make the support structure itself brittle by cooling down with fan at maximum speed. In my case, the default fan speed is not 100% making the support structure more stiff than it have to suppose to be

A new parameter for fan speed and supports I think that would be useful.

user32rtu, 2021-02-18:
To make the supports brittle, it is enough to make the width of the support line slightly less than the diameter of the nozzle. For a 0.4 mm nozzle, you can set the width to 0.35 mm. This works well.

neophyl, 2021-02-18:
Also what contact z distance are you using ?  Anything less than 0.2 makes them very difficult to remove.

user32rtu, 2021-02-18:
It is necessary to experimentally select the Z gap for plastic.

tecnopolis-ve, 2021-02-18:
Hello, 

let me clarify: actually in my current situation I'm able to break apart support from printed part, but sometimes the support structure is so massive that it would be easier if I'm able to shatter it into smaller pieces.

That's why I opened this feature request; the layers of the support structure itself are really well fused each other due my current settings. Additionaly, I'm using underextrusion for support structures (75% width from nozzle diameter)

Thanks

user32rtu, 2021-02-22:
Hello. I found a way to help you. Add to "Printer Settings-> Custom G-code-> Between extrusion role change G-code" 
{if extrusion_role == "SupportMaterial" or extrusion_role == SupportMaterialInterface "} M106 S255 
  {else} M107 
{endif}.
This will turn on the fan at 100% on the supports.

supermerill, 2021-02-22:
> I think that 100% fan speed for supports would be a fix.
 
 can you test it with 100% to see if it's really useful?

tecnopolis-ve, 2021-02-22:
Hello @supermerill 

Nice to hear from you again!

I'll test the suggestion given by @user32rtu, but as I've explained; my pla settings are meant to generate strong pieces by reducing general cooling except for overhangs/bridges/external shell, and, most of the time, the support structure is not cooling enough, making it pretty stiff to break with bare hands.

Thanks

Dantoes, 2022-04-09:
I'm having issues with hard to remove supports as well. After a bit of tuning, the top side of the support structure can be easily removed from the object. But if the supports have to be created on the object, they are fused onto it and are extremely hard to remove. The z-offsets for bottom and top supports are equal. The only difference I noticed, is that the bottom support plates ("support material interface") repect the "max fan speed", while the object layer on top of the supports make use of the "bridges fan speed" setting. Maybe, the feature "support material interface" should use the bridge fan speed as well.

closed 2024-02-11 by supermerill