supermerill/SuperSlicer #939: Speedup time issue

qwewer0, 2021-02-02:
### Version
2.3.55.2

### Operating system type + version
Windows 10, 10.0.19042 Build 19042

### Behavior

1. _Speedup time_ (`fan_speedup_time`) always taken into account (unless zero), even if _Only for overhangs_ (`fan_speedup_overhangs`) is selected.
2. Both positive and negative values for _Speedup time_ (`fan_speedup_time`) has the same effect.

### Steps to reproduce
1. Generate a calibrating cube (or load a file)
2. Switch to _Fan speed_ view
3. See the fan speed differences

**Expected behavior**

1. By selecting _Only for overhangs_ (`fan_speedup_overhangs`) the _Speedup time_ (`fan_speedup_time`) should only be used for overhangs.
2. For _Speedup time_ (`fan_speedup_time`) a negative value should cause the fan speed change to occur sooner, and a positive value should cause the fan speed change to occur later.

**Actual behavior**
1. By selecting _Only for overhangs_ (`fan_speedup_overhangs`) the _Speedup time_ (`fan_speedup_time`) is used everywhere not just on overhangs.
2. For _Speedup time_ (`fan_speedup_time`) a negative value causes the fan speed change to occur sooner (as it should), but a positive value has the same effect.

**Screenshots**

**Speedup time = 0** (`fan_speedup_time`)
[image]

**Speedup time = -1** (`fan_speedup_time`)
[image]

**Speedup time = 1** (`fan_speedup_time`)
[image]

#### Project File (.3MF) where problem occurs
[Shape-Box-3mf.zip](https://github.com/supermerill/SuperSlicer/files/5913087/Shape-Box-3mf.zip)

0x4d4e, 2021-03-15:
I can confirm that 'only for overhangs' is not taken into account. In addition, I noticed inconsistent behavior with fan_speedup_time/overhang detection which I think is related to this as well:

3s fan speedup (only overhangs true) - some overhangs are not 'detected' at all, front and backside differ in similar sections

Front
[image]

Back
[image]

1s fan speedup (only overhangs true) - different 'pattern' - still differences between front and back
[image]
[image]

In comparison, with fan speedup off:
[image]

supermerill, 2021-03-18:
I'll rewrite it entirely in the new gcode writer later this year.

gpread, 2021-08-29:
2.3.56.8

There is still a problem :-(  but its very close to working as intended/requested it seems! :-)

When "only for overhangs" is checked, currently its correctly adding the delay/advance (+/-) for only a single feature type but to the wrong one, its doing it for "Internal Bridge Infill" rather than "Overhang Perimeter"

qwewer0, 2021-11-07:
Fixed.

closed 2021-11-07 by qwewer0