supermerill/SuperSlicer #940: macOs: Serif Font in buttons

ZhuJo, 2021-02-03:
### Version
2.3.55.2

### Operating system type + version
macOS 11.2 (same on older)

### 3D printer brand / version + firmware version (if known)
irrelevant

### Behavior
* On macOs, some of the buttons have Serif font instead of default, sans serif one. It looks ugly. Prusaslicer doesn't have this issue
[image]

Stout001, 2021-02-19:
Strange. I use 3 different MacOS versions on 4 computers and the font is constant and not broken.

ZhuJo, 2021-02-20:
Well it happens on two different computers. Tried to clean font cache, nope. 
New user account, nope.

supermerill, 2021-02-22:
fixed by the merge

closed 2021-06-04 by supermerill