supermerill/SuperSlicer #945: "Only one perimeter on Top surfaces" option leading to loss of vertical shells when minumum vert. shells is wider than distance between walls

scottmudge, 2021-02-04:
### Version
Latest - 2.3.55

### Operating system type + version
Win64

### 3D printer brand / version + firmware version (if known)
MK3S+

### Behavior
When "Only one perimeter on Top surfaces" option is enabled, there is occasional loss of vertical shells of numerous parts, where the distance between the walls is **smaller than** the minimum required number of vertical shells.

I have set minimum number of vertical shells to 6 (12 total width), which is larger than the width of the region where the vertical shells are lost. Instead of simply filling the region completely, it removes all vertical shells:

[image]

If I uncheck this option or reduce the vert. shell minimum to something like 4, this problem does not happen.

#### Project File (.3MF) where problem occurs
[loss_of_vert_shells.zip](https://github.com/supermerill/SuperSlicer/files/5927593/loss_of_vert_shells.zip)

qwewer0, 2021-02-05:
The issue is reproduceable on 2.3.55.4

BenWhite27, 2021-02-23:
Still reproduceable on 2.3.55.5, also affects gapfill.

supermerill, 2021-03-18:
I'll revert the algo to the version before #733, with an hidden modifier-switch for when the issue arise again.

closed 2021-06-04 by supermerill