supermerill/SuperSlicer #948: Z-Lift modifier doesn't work.

user32rtu, 2021-02-06:
### Version
_Version of SupserSlicer used goes here, Use `About->About SuperSlicer`_

### Operating system type + version
Windows 10 Pro 20H2

### 3D printer brand / version + firmware version (if known)
Sapphire Pro

### Behavior
I am trying to avoid moving the nozzle across the top filled surfaces as the nozzle leaves scars on them.
I am using the height range modifier with a Z-lift override. But no value for Z-lift has any effect.
This only works when Lift Z is set to a non-zero positive value in the filament profile. But in this case Z Lift works on all layers, and I don't need it.

#### Project File (.3MF) where problem occurs
[image]
[image]
[Left part.zip](https://github.com/supermerill/SuperSlicer/files/5937429/Left.part.zip)

qwewer0, 2021-02-06:
@user32rtu Make sure that you don't have **Lift Z** set to **0** in **Filament Overrides**.
Don't forget to close the issue if the problem is solved.

closed 2021-02-07 by user32rtu

qwewer0, 2021-02-07:
@user32rtu Did you find a solution to the problem?