supermerill/SuperSlicer #949: The meaning of "project" is inconsistent

JohanAR, 2021-02-06:
### Version
2.3.55.4

### Operating system type + version
Xubuntu 20.04.2 LTS

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
* "New Project" resets the plater, but does not touch the settings. Any modified settings are kept.
* "Open Project..." replaces both plater *and* settings. Any modified settings are silently discarded.
* Undo after opening a new project reverts the plater contents, but does not revert the loaded settings.

The problem is that a "project" sometimes means only models on the plater, and sometimes both plater contents and settings.

### Proposed solution
* Have "project" always refer to a combination of plater (which may or may not contain models) and settings.
* If it should be possible to undo an "open project", it needs to revert both plater contents and settings together. Personally I don't think undoing switching projects is necessary, it is not a feature I expect. It would of course be nice to have, but I think it might add a lot of complexity to the code.
* If it should *not* be possible to undo an "open project", it should present the user with the unsaved changes dialog, giving them the option to save/transfer/discard changes.

#### Project File (.3MF) where problem occurs
N/A

Redferne, 2021-02-09:
Yes, I have this problem as well. I'm afraid of loading old projects since the current settings are lost forever.

azaz44, 2021-03-06:
I have this problem too, and had a situation where it is really dangerous. I printed PLA at 250 degrees because of that.

Steps to reproduce:
1. Use menu Calibration ->Filament Temperature Calibration to generate a temp tower
2. Slice it
3. Now, you don't need it anymore, and want to print something new
4. Delete the tower from the plate
5. Now try to make "New Project" -- you can't. It is greyed out.
6. Add any STL to the plate and slice.
7. Now you thing you are fine printing it. But actually your code contains G-Codes for temperature switching left from the previous project.

supermerill, 2021-03-18:
And currently, when there is only one object, "delete" (one) is not the same as "delete all" because the second one is in fact "new project" ....

Also, ctrl+z may restore the previous parts but not the previous settings...

Have to do something here.

Ideally, a "recent projects" list in the menu may be useful.

supermerill, 2021-03-18:
> About "delete all" -- I think user expectation would be just to remove parts, not change settings. But yes, it gets tricky here... because why would you delete all parts if you're not starting a new project...

I think I will change the pop-up from "yes" "no"(cancel) "cancel" to "Remove objects" "New project" "Cancel"

grantemsley, 2021-12-30:
Along these lines, when opening a .3mf file that includes settings, it would be really nice if I got a prompt like:

    Use settings from this project?
    ---------------------------         ----------------------------------
    |Yes, use project settings|         |No, keep current slicer settings|
    ---------------------------         ----------------------------------

Clicking yes would import all the settings from the project, overwriting your current slicer settings.

No would keep your existing print, filament and printer settings - but ideally still import any per-object settings that were saved in the file.

supermerill, 2022-01-04:
understood

I have to create some flow chart documentation, so I can keep track on what should be doing what.