supermerill/SuperSlicer #951: feature request: arc welder integration

thebrelon, 2021-02-08:
Would it be possible to integrate arc welder to super slicer, with adjustable settings like resolution, etc... ?

CCS86, 2021-02-27:
I think it would be a great feature.

supermerill, 2021-03-18:
Will do it after / inside the gcode export rewrite.

bttguy, 2021-05-24:
+1 from me, cant wait this feature. I know there are workarounds, but looks like SuperSlicer is the slicer for a proper integration :)

BrianGilbert, 2021-11-08:
just came looking to see if this is possible yet

CCS86, 2021-11-09:
Just something to keep in mind guys. Your firmware might not support arcs as directly as you would hope. I was really hoping that reprap firmware on my duet board would do better with arcs, but in reality it just decomposes arcs into line segments of a fixed length, and actually made faceting worse compared to linear code straight out of my slicer.

At least with super slicer you have direct control over the size of the line segments describing your arcs and you can tune them for the speeds you print at to be as dense as possible without exceeding the G1 per second limit of your board.

adamrumjahn, 2022-02-24:
+1 for this!

Treintjes, 2022-03-17:
@supermerill have we made forward progress on adding arcwelder?

supermerill, 2022-03-17:
@Treintjes 
@lordofhyphens has added the arcwelder lib repo into the code base, but I don't know much more.

P-tite-tete, 2022-03-21:
+1 for this!

digitalentropy, 2022-04-28:
> For anyone wondering I was able to download arcwelder.exe and then use it via output option via directory so when I hit slice arcwelder will post process

This is a solid tip, and hopefully it will eventually be integrated such that we can modify settings within superslicer

adamrumjahn, 2022-04-28:
Can you please show what you did?  I checked the Output options in the
Print Tab but I don't see where you can execute Arcwelder.exe.

--
*Adam Rumjahn*
Managing Director

*Orion Additive Manufacturing GmbH*
  - Office & Shipping Address / *Büro & Lieferadresse:*
    Orion Additive Manufacturing GmbH
    c/o Start-A-Factory (Fraunhofer IZM)
    Gustav-Meyer-Allee 25, Treppe 17/3, 13355 Berlin, Germany
    (directions <https://g.page/OrionAM?share>:map
<https://drive.google.com/open?id=1ap18P5czylCy5f65s7pE6nWRgFkZH4fq>)
*  - Billing Address / Rechnungsadresse:*
    Bandelstrasse 24, 10559 Berlin, Germany
*Phone:* +49 304 6403 8403
*Mobile:* +49 176 6701 6284
*Social Media:* Linkedin
<https://www.linkedin.com/in/adam-rumjahn-46b95433/>
*Email:* ***@***.***
*Web:* www.orion-am.com
*Handelsregister:* HRB 201874 B
*Geschäftsführer:* Adam Rumjahn
*USt-IdNtNr:* DE 320 942 964
*EORI #: *DE372245255125324

[image: Mailtrack]
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality11&>
Sender
notified by
Mailtrack
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality11&>
28/04/22,
16:03:32

adamrumjahn, 2022-04-28:
Awesome, thanks @supermerill .  I got it up and running now.  I'm looking forward to testing it out.

Lite5h4dow, 2022-06-01:
want to bump this. i this fixes so many print issues i have rn.

CCS86, 2022-06-02:
> want to bump this. i this fixes so many print issues i have rn.

What issues?

Lite5h4dow, 2022-06-02:
> What issues?

I print fast and with superslicers current settings I get a lot of stuttering and ribbing on round edges if I don't run them through arc welder first.i don't use octoprint either, so I cand use the arc welder octoprint plugin, so I have to do it manually every time and it's tedious. If I don't I get stutter ribbing.

StephenRC, 2022-06-02:
Just put it in the Post-Processing Script, under Print Settings->Output 
options.

On 6/2/2022 7:21 AM, LiteLotus wrote:
>
>     What issues?
>
> I print fast and with superslicers current settings I get a lot of 
> stuttering and ribbing on round edges if I don't run them through arc 
> welder first.i don't use octoprint either, so I cand use the arc 
> welder octoprint plugin, so I have to do it manually every time and 
> it's tedious. If I don't I get stutter ribbing.
>
> Message ID: ***@***.***>
>

-- 
This email has been checked for viruses by AVG.
https://www.avg.com

wolfallein, 2022-12-04:
+1 for the integration into SuperSlicer