supermerill/SuperSlicer #952: Adaptive Layer Height Slicing does not execute on simple model

tobor8man, 2021-02-08:
### Version
2.3.55.2
2.3.55.4

### Operating system type + version
Windows 10 Home Version 2004 OS build 19041.746

### 3D printer brand / version + firmware version (if known)
Anet A8
Anycubic Kossel Linear Plus

### Behavior
* _Describe the problem_
Slicing does not respect Adaptive Layer Height settings. Constant layer height gcode is returned.

* _Steps needed to reproduce the problem_
Import STL
Select Adapative Layer Height control
Click Adaptive Layer Height to preview layers (min 0.05, max 0.3 defined in settings)
Click Slice Now
Preview shows constant layer height as defined in printer setting

* _Expected Results_
Gcode with varied layer heights

 
* _Actual Results_

[image]
[image]

#### Project File (.3MF) where problem occurs

[Hemisphere (MM).zip](https://github.com/supermerill/SuperSlicer/files/5947275/Hemisphere.MM.zip)

qwewer0, 2021-02-09:
I too have that issue with your 3mf file, but not with my own settings. (2.3.55.4)
[Shape-Sphere.zip](https://github.com/supermerill/SuperSlicer/files/5949498/Shape-Sphere.zip)

[image]
[image]
[image]

tobor8man, 2021-02-09:
> I too have that issue with your 3mf file, but not with my own settings. (2.3.55.4)

Can you try with the source STL (attached) and min/max layer height of 0.05/0.3 respectively. The remaining adaptive steps are default.
[Hemisphere (MM) STL.zip](https://github.com/supermerill/SuperSlicer/files/5950943/Hemisphere.MM.STL.zip)

In the meantime I'll try your settings. Cheers

qwewer0, 2021-02-09:
> Can you try with the source STL (attached) and min/max layer height of 0.05/0.3 respectively. The remaining adaptive steps are default.
> [Hemisphere (MM) [...]

Tried the attached stl with min/max layer height of 0.05/0.3, but it is still working for me. So either there is a setting that intentionally suppresses the adaptive layer setting or it is unintentional, and so a bug.

tobor8man, 2021-02-09:
I tried your file. Something in it does not let me change the min/max limits to 0.05/0.3 mm. Your file has them set at 0.08/0.28 but when I try to change it locks to 0.04/0.32. Looks like its locked to multiples of 0.04, inherited from the Ender config maybe. Not sure I can do much more at this stage, Pretty new to SuperSlicer. Might just be a quite specific failure not worth digging into. Thanks for looking at the issue.

qwewer0, 2021-02-09:
> I tried your file. Something in it does not let me change the min/max limits to 0.05/0.3 mm. Your file has them set at 0.08/0.28 but when I try to change it locks to 0.04/0.32. Looks like its locked [...]

Change the Z full step (z_step).

supermerill, 2021-03-18:
As @qwewer0 said, It may be because the Z full step is set.

closed 2021-06-04 by supermerill