supermerill/SuperSlicer #953: Bridge infill height issue

qwewer0, 2021-02-09:
### Version
2.3.55.4

### Operating system type + version
Windows 10, 10.0.19042 Build 19042

### Description
Changing the bridge flow (`bridge_flow_ratio`) also changes the layer height of the bridge infill in relation to the nozzle diameter.
e.g. if the bridge flow is 100%, then the bridge infill layer height will be equal to the nozzle diameter.

`bridge_flow_ratio = 100%`
[image]

`bridge_flow_ratio = 100%`
[image]

`bridge_flow_ratio = 50%`
[image]

`bridge_flow_ratio = 200%`
[image]

### Expected behavior
Changing the bridge flow shouldn't change the layer height of the bridge infill. (or any)

### Actual behavior
Changing the bridge flow changes the layer height of the bridge infill in relation to the nozzle diameter.

#### Project File (.3MF) where problem occurs
[Shape-Box.zip](https://github.com/supermerill/SuperSlicer/files/5951229/Shape-Box.zip)

neophyl, 2021-02-09:
The way Prusa/super slicer do bridges is it always assumed that the printed layer height is the same diameter as the extrusion width/nozzle size as that layer is not being squished at all. As such it can only be that size. You increase flow out the nozzle then the printed diameter also increases. 
There are numerous issues on the Prusa GitHub relating to this. Might even be some here, without doing a search I can’t remember. 
Not commenting on if this assumption is valid or not, just an explanation of why it is that way.

qwewer0, 2021-02-09:
> The way Prusa/super slicer do bridges is it always assumed that the printed layer height is the same diameter as the extrusion width/nozzle size as that layer is not being squished at all. As such [...]

If the slicer would use the extrusion width instead of the nozzle diameter, then there wouldn't be any problem, but that isn't the case right now.
In the images the larger nozzle difference stands out more, but the problem is still there if the nozzle is smaller than the used extrusion widths, so the bridge layer height would be smaller than expected.

The layer height difference will throw off the users when they see it, and no logical way to change it, so maybe slicer should have a bridge extrusion width setting to avoid this kind of unintentional value change, but either way the slicer definitely shouldn't use the nozzle as an extrusion width base value.

ajalberd, 2021-02-22:
I'm seeing bridge extrusions way below where they should be in the model. Why is this? Any way to fix? 
[image]

The blue bridges should be completely gone - the orange is supposed to be a smooth curve and those 5 bridge lines ruin the slope.

supermerill, 2021-02-22:
problem is the slicer slice the layers, and then decided what should be bridge flow or not. So these areas will "drip" on the previous layer(s).
Have to modify the entire slicing & bridge of the slicer to counteract that.
I tried to began but it's a bit like rewriting the most basic foundation of the software, so it's a very big task.

neophyl, 2021-02-22:
Please read my above post. As the layers are assumed to be the same height as the nozzle width/extrusion width ie 0.4/0.45 then those bridging layers are thicker.  As such the render engine displays them as that thickness.  They also get printed that thickness so the render is actually accurate. 
You can turn off bridge detection which will make them normal layer thickness but of course you don’t get bridge extrusion rates or bridge fan speed etc then. It also doesn’t turn off the bridge infill as shown as light blue in your screen shot. At the moment there’s nothing you can do about that.

supermerill, 2021-03-18:
And using somthign else than bridge/overhang flow when there is nothing under it is useless because your extrusion irl won't be the fatty rectangle it display but a circle! By doing that you just lost the visual cue that may help you.

qwewer0, 2021-03-18:
> And using something else than bridge/overhang flow when there is nothing under it is useless because your extrusion irl won't be the fatty rectangle it display but a circle! By doing that you just lost the visual cue that may help you.

Maybe a setting to turn that visual representation off?

supermerill, 2021-03-19:
> Maybe a setting to turn that visual representation off?

Turn the bridge representation into something else? I won't do it, have bugs to resolve and more important features to implement.

Btw, there's the real issue that bridge areas are created too low.

qwewer0, 2021-03-19:
> Turn the bridge representation into something else?

I meant by that to have the bridges and overhang to either shown as it is right now, or as the layer height around it. (It could inherit the shown layer height from the line that it is connected to, or something like that, as I have no ide how this could work...)

> Btw, there's the real issue that bridge areas are created too low.

Didn't know that, maybe my ide should wait until that fix.

@supermerill Thank you for everything!

qwewer0, 2021-09-08:
When disable thick_bridges feature is merged from PrusaSlicer, this issue can be closed.

supermerill, 2022-03-23:
in nightly, you can select other way of computing the bridge flow, should be easier to tune.

closed 2022-03-29 by qwewer0