supermerill/SuperSlicer #954: Unnecessarily M486 in file

qwewer0, 2021-02-09:
### Version
2.3.55.4

### Operating system type + version
Windows 10, 10.0.19042 Build 19042

### Description
The slicer adds unnecessarily `M486` (`M486 T1`, `M486 S0`, `M486 S-1`) in the .gcode file.

2.3.55.2 doesn't add `M486` to the .gcode file.
[M486](https://marlinfw.org/docs/gcode/M486.html)

### Expected behavior
Don't add M486 to the .gcode file if it isn't asked.

### Actual behavior
M486 is added to the .gcode file.

#### Project File (.3MF) where problem occurs
[Shape-Box.zip](https://github.com/supermerill/SuperSlicer/files/5951338/Shape-Box.zip)

DaStivi, 2021-02-24:
experienced the same issue right now... Problem is that my BTT TFT cannot interpret the cancle object command and genereates am gcode command error each time the code is processed...

basically i understand the idea of M486 in multipart prints, to be able to cancle one part prints if it fails... so nice too see this addition in superslicer, but there should be either ah option too turn off or even ignore it on single part models alltogheter... double checked my multipart/sequential printing options in superslicer options but its all disabled

qwewer0, 2021-02-24:
The problem still exist on 2.3.55.5

DaStivi, 2021-02-25:
@paukstelis @supermerill   i would just like raise the awarness here... (it might looks also connected to the mac crashing issues?!)

paukstelis, 2021-02-25:
I hadn't even noticed that M486s had made it in! I think an additional tick box in the Output options for this would make sense.

DaStivi, 2021-03-01:
I've found the options...  disable "Label Objects" does the job to not write the M486 Code into the gcode file anymore.. 
funny thing, as i just found out, as i have setup my octoprint-pi now and installed the cancelobject plugin, just to find on the cancelobject plugin's github page how the setting is called in prusa/superslicer 🙈🤦‍♂️

https://github.com/paukstelis/Octoprint-Cancelobject#prusaslicersuperslicer
[image]

qwewer0, 2021-03-01:
> I've found the options... disable "Label Objects" does the job to not write the M486 Code into the gcode file anymore..
> funny thing, as i just found out, as i have setup my octoprint-pi now and [...]

Thank you!

closed 2021-03-01 by qwewer0

supermerill, 2021-06-03:
Currently M486 is only added to marlin firmware. I'll also enable it to reprapfirmware, as they added the support for that.

oliverhbailey, 2022-12-09:
GCODE was developed to be CNC Machine Agnostic 60 years ago.  There are a list of these codes that have sustained without fault for 60 years because they were designed by professional engineers that know the mechanics and physics of motion control.  The M486 command is not a part of standard GCODE and therefore is not recognized by a standard controller built to GCODE control standards.  As with most non-standard GCODEs that are sent to a standard GCODE compliant controller, they throw and error message and Pause the controller on that instruction.  Unfortunately Marlin is not the standard and firmware that is GCODE standard compliant will not properly interpret the M486 instruction.  This should be remedied by having a checkbox that allows standard GCODE commands only or Marlin specific GCODE check box in the slicers.  Forcing users who know GCODE to track down Marlin specific GCODES by default is a bad idea from the perspective of gaining users who are using machines where M486 commands are not a part of the firmware.

paukstelis, 2023-01-10:
> GCODE was developed to be CNC Machine Agnostic 60 years ago. There are a list of these codes that have sustained without fault for 60 years because they were designed by professional engineers that [...]

That was a fun read. Good to know that there are still plenty of "if you aren't doing it my way, it's wrong" types out there.

oliverhbailey, 2023-01-10:
It has nothing to do with doing it "my way".  It means that G-code was
designed for physical motion control.  Anything custom can, and should be
handled as a macro.  And macros can be stored in firmware.  It also means
the person writing the program read the published standad syntax before
writing or changing it.  Depending the knowledge base of the person reading
that comment, it would be interpreted as informative, or not understood,
putting it in the entertaining category.

Glad you were "entertained".