supermerill/SuperSlicer #955: Crashes on slicing model

andreyoleynik, 2021-02-10:
### Version
2.3.55.2

### Operating system type + version
MacOS Big Sur 11.1 (20C69)

### 3D printer brand / version + firmware version (if known)
Longer LK4Pro, Marlin

### Behavior
* _Describe the problem_
On press slice button it crashes

* _Steps needed to reproduce the problem_
Load stl file, press slice button

* _Expected Results_
Normal slice
* _Actual Results_
Crash
  * _Screenshots from __*SuperSlicer*__ preview are preferred_

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`File -> Save` / `Save Project`)_

[Archive.zip](https://github.com/supermerill/SuperSlicer/files/5954967/Archive.zip)

andreyoleynik, 2021-02-10:
This is crash log

Process:               SuperSlicer [13018]
Path:                  /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer
Identifier:            com.merill.slic3r/
Version:               SuperSlicer SuperSlicer_2.3.55.2 (SuperSlicer_2.3.55.2)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           SuperSlicer [13018]
User ID:               501

Date/Time:             2021-02-10 02:15:10.281 +0200
OS Version:            macOS 11.1 (20C69)
Report Version:        12
Bridge OS Version:     5.1 (18P3030)
Anonymous UUID:        ED844C98-B164-A97A-1C60-07C25A9C4EFB

Sleep/Wake UUID:       E7B91EBE-5F35-4C80-ACC1-EE7830DCEE7C

Time Awake Since Boot: 220000 seconds
Time Since Wake:       58000 seconds

System Integrity Protection: enabled

Crashed Thread:        10  slic3r_tbb_4

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000010
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [13018]

VM Regions Near 0x10:
--> 
    __TEXT                      101e2f000-103b5f000    [ 29.2M] r-x/r-x SM=COW  /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer

Thread 0:: slic3r_main  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff20338e7e mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff203391f0 mach_msg + 60
2   com.apple.CoreFoundation      	0x00007fff20465bf7 __CFRunLoopServiceMachPort + 316
3   com.apple.CoreFoundation      	0x00007fff204642ca __CFRunLoopRun + 1315
4   com.apple.CoreFoundation      	0x00007fff204636ce CFRunLoopRunSpecific + 563
5   com.apple.HIToolbox           	0x00007fff286e86d0 RunCurrentEventLoopInMode + 292
6   com.apple.HIToolbox           	0x00007fff286e84cc ReceiveNextEventCommon + 709
7   com.apple.HIToolbox           	0x00007fff286e81ef _BlockUntilNextEventMatchingListInModeWithFilter + 64
8   com.apple.AppKit              	0x00007fff22c80de9 _DPSNextEvent + 883
9   com.apple.AppKit              	0x00007fff22c7f5af -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1366
10  com.apple.AppKit              	0x00007fff22c71b0a -[NSApplication run] + 586
11  com.merill.slic3r/            	0x000000010359ddde wxGUIEventLoop::OSXDoRun() + 174
12  com.merill.slic3r/            	0x00000001036faea7 wxCFEventLoop::DoRun() + 39
13  com.merill.slic3r/            	0x000000010363455f wxEventLoopBase::Run() + 143
14  com.merill.slic3r/            	0x00000001035fca5f wxAppConsoleBase::MainLoop() + 191
15  com.merill.slic3r/            	0x00000001035fc67c wxAppConsoleBase::OnRun() + 28
16  com.merill.slic3r/            	0x000000010330f74b wxAppBase::OnRun() + 59
17  com.merill.slic3r/            	0x000000010353f549 wxApp::OnRun() + 41
18  com.merill.slic3r/            	0x000000010366f32c wxEntry(int&, wchar_t**) + 300
19  com.merill.slic3r/            	0x000000010366f4fb wxEntry(int&, char**) + 59
20  com.merill.slic3r/            	0x0000000102699caf Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) + 223
21  com.merill.slic3r/            	0x0000000101e3b4fb Slic3r::CLI::run(int, char**) + 16875
22  com.merill.slic3r/            	0x0000000101e40047 main + 439
23  libdyld.dylib                 	0x00007fff20388621 start + 1

Thread 1:: com.apple.NSEventThread
0   libsystem_kernel.dylib        	0x00007fff20338e7e mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff203391f0 mach_msg + 60
2   com.apple.CoreFoundation      	0x00007fff20465bf7 __CFRunLoopServiceMachPort + 316
3   com.apple.CoreFoundation      	0x00007fff204642ca __CFRunLoopRun + 1315
4   com.apple.CoreFoundation      	0x00007fff204636ce CFRunLoopRunSpecific + 563
[... 774 lines cut ...]
Foundation                          48K        2 
Kernel Alloc Once                    8K        1 
MALLOC                           335.4M      107 
MALLOC guard page                   32K        8 
MALLOC_NANO (reserved)           384.0M        1         reserved VM address space (unallocated)
OpenGL GLSL                        256K        3 
STACK GUARD                       56.1M       33 
Stack                             80.3M       34 
VM_ALLOCATE                       3072K       72 
__CTF                               759        1 
__DATA                            29.8M      379 
__DATA_CONST                      21.8M      223 
__DATA_DIRTY                       935K      133 
__FONT_DATA                          4K        1 
__GLSLBUILTINS                    5176K        1 
__LINKEDIT                       514.0M       11 
__OBJC_RO                         60.5M        1 
__OBJC_RW                         2452K        2 
__TEXT                           391.3M      377 
__UNICODE                          588K        1 
mapped file                      242.4M       40 
shared memory                       44K        5 
===========                     =======  ======= 
TOTAL                              2.4G     1844 
TOTAL, minus reserved VM space     2.0G     1844

pavelbrylov, 2022-02-13:
FYI: Getting similar on 2.3.57.10
```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               SuperSlicer [2796]
Path:                  /Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer
Identifier:            com.merill.slic3r/
Version:               SuperSlicer SuperSlicer_2.3.57.10_2022-02-03 (SuperSlicer_2.3.57.10_2022-02-03)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2022-02-13 22:39:08.1905 +0100
OS Version:            macOS 12.1 (21C52)
Report Version:        12
Bridge OS Version:     6.1 (19P647)
Anonymous UUID:        248A0CAA-95B5-30F2-1AE0-D30DB2AB4335

Sleep/Wake UUID:       EBBC0FF8-917A-436D-9A91-A45224710832

Time Awake Since Boot: 2300000 seconds
Time Since Wake:       119715 seconds

System Integrity Protection: enabled

Crashed Thread:        7  slic3r_BgSlcPcs

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [2796]

VM Region Info: 0 is not in any region.  Bytes before following region: 4370120704
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      1047ac000-106544000    [ 29.6M] r-x/r-x SM=COW  ...S/SuperSlicer

Thread 0:: slic3r_main Dispatch queue: CGImageProviderCallbackQueue
0   libsystem_kernel.dylib        	    0x7ff81a8941c2 kevent_id + 10
1   libdispatch.dylib             	    0x7ff81a7376ce _dispatch_kq_poll + 164
2   libdispatch.dylib             	    0x7ff81a738180 _dispatch_event_loop_wait_for_ownership + 521
3   libdispatch.dylib             	    0x7ff81a72638c __DISPATCH_WAIT_FOR_QUEUE__ + 279
4   libdispatch.dylib             	    0x7ff81a725fe5 _dispatch_sync_f_slow + 170
5   CoreImage                     	    0x7ff823b436ef CI::MetalContext::render_node(CI::TileTask*, CI::ProgramNode*, CGRect const&, CGRect const&, void const**, __IOSurface**, unsigned long) + 269
6   CoreImage                     	    0x7ff823d442ec CI::MetalContext::render_root_node(CI::TileTask*, CI::ProgramNode*, CGRect const&, void () block_pointer, void () block_pointer) + 340
7   CoreImage                     	    0x7ff823d45aea CI::Context::recursive_render(CI::TileTask*, CI::roiKey const&, CI::Node*, bool) + 2184
8   CoreImage                     	    0x7ff823b4325f CI::Context::render(CI::ProgramNode*, CGRect const&) + 109
9   CoreImage                     	    0x7ff823d8a748 invocation function for block in CI::image_get_bitmap(CI::Context*, CI::Image*, CGRect, CGColorSpace*, CI::Bitmap*, CI::RenderDestination const*) + 1188
10  CoreImage                     	    0x7ff823b3c8cf CI::recursive_tile(CI::RenderTask*, CI::Context*, CI::RenderDestination const*, char const*, CI::Node*, CGRect const&, CI::PixelFormat, CI::swizzle_info const&, CI::TileTask* (CI::ProgramNode*, CGRect) block_pointer) + 3569
11  CoreImage                     	    0x7ff823b3b5df CI::tile_node_graph(CI::Context*, CI::RenderDestination const*, char const*, CI::Node*, CGRect const&, CI::PixelFormat, CI::swizzle_info const&, CI::TileTask* (CI::ProgramNode*, CGRect) block_pointer) + 444
12  CoreImage                     	    0x7ff823b3b121 CI::image_get_bitmap(CI::Context*, CI::Image*, CGRect, CGColorSpace*, CI::Bitmap*, CI::RenderDestination const*) + 1706
13  CoreImage                     	    0x7ff823b3aa1e CI::RenderToBitmap::render(CI::Image*, CI::Context*) const + 524
14  CoreImage                     	    0x7ff823b3a2fb -[CIContext(CIRenderDestination) _startTaskToRender:toDestination:forPrepareRender:forClear:error:] + 3429
15  CoreImage                     	    0x7ff823b3958f -[CIContext(CIRenderDestination) startTaskToRender:toDestination:error:] + 31
16  CoreImage                     	    0x7ff823b393bf -[CIContext(CIRenderDestination) startTaskToRender:fromRect:toDestination:atPoint:error:] + 170
17  CoreImage                     	    0x7ff823bb9adc invocation function for block in CI::copyImageBlockSetOptsCallback(void*, CGImageProvider*, CGRect, CGSize, __CFDictionary const*) + 636
18  libdispatch.dylib             	    0x7ff81a718cc9 _dispatch_client_callout + 8
19  libdispatch.dylib             	    0x7ff81a72613f _dispatch_lane_barrier_sync_invoke_and_complete + 60
20  CoreImage                     	    0x7ff823b3884b CI::copyImageBlockSetOptsCallback(void*, CGImageProvider*, CGRect, CGSize, __CFDictionary const*) + 1140
21  CoreGraphics                  	    0x7ff81fc3dc5f imageProvider_retain_data + 77
22  CoreGraphics                  	    0x7ff81fc3dbd6 CGDataProviderRetainData + 75
23  CoreGraphics                  	    0x7ff81fc3da96 CGAccessSessionCreate + 98
24  CoreGraphics                  	    0x7ff81fc0d1e9 CGDataProviderCopyData + 191
25  CoreGraphics                  	    0x7ff81fc3a403 CGImageGetDataProviderInternal + 89
[... 599 lines cut ...]
USB Device: USB Audio CODEC
USB Device: USB Receiver
USB Device: Anker USB-C Hub Device
USB Device: USB3.0 Hub
USB Device: USB3.0 Card Reader
USB Device: USB30Bus
USB Device: USB3.1 Hub
USB Device: hub_device
USB Device: LG UltraFine Display Camera
USB Device: USB2.1 Hub
USB Device: hub_device
USB Device: USB Controls
USB Device: USB Audio
USB Device: T2Bus
USB Device: Touch Bar Backlight
USB Device: Touch Bar Display
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 47.5
Thunderbolt Bus: MacBook Pro, Apple Inc., 47.5
Thunderbolt Device: UltraFine 5K, LG Electronics, 3, 9.4
```

supermerill, 2022-02-13:
please use spoilers for logs
https://gist.github.com/jbsulli/03df3cdce94ee97937ebda0ffef28287

for the 2.3.55 issue, please update to 2.3.57
for the 2.3.57.10 issue, there is a good chance it's the brim bug, I just release the 2.3.57.11 to fix it. Sorry for it.