supermerill/SuperSlicer #956: Crash when Slicing a specific model

Minni1337Me, 2021-02-10:
### Version
2.3.55.2

### Operating system type + version
Windows 10 latest Version

### Behavior
* _Describe the problem_

Slicer crashes when Slicing the model

* _Steps needed to reproduce the problem_
I load the model, press Slice, SuperSlicer Crashes
* _Expected Results_
I get a STL file
* _Actual Results_
Super Slicer Crashes

#### Project File (.3MF) where problem occurs

[honeycomb_angled_nespressodispenser-v50.zip](https://github.com/supermerill/SuperSlicer/files/5961777/honeycomb_angled_nespressodispenser-v50.zip)

qwewer0, 2021-02-10:
It doesn't crashes for me on 2.3.55.4

Minni1337Me, 2021-02-10:
where can i get 2.3.55.4? do I have to compile it myself? For me it says 2.3.55.2 is the latest version...

Minni1337Me, 2021-02-10:
Okay, Checked again. 18% infill I get a crash, 10% infill I dont get a crash...

qwewer0, 2021-02-10:
> where can i get 2.3.55.4? do I have to compile it myself? For me it says 2.3.55.2 is the latest version...

You can find it in the actions tab

Minni1337Me, 2021-02-10:
> > where can i get 2.3.55.4? do I have to compile it myself? For me it says 2.3.55.2 is the latest version...
> 
> You can find it in the actions tab

thank you very much! I´ll try.

closed 2021-03-18 by supermerill