supermerill/SuperSlicer #958: Top solid layer travels are not optimized properly.

burner-, 2021-02-12:
### Version
2.3.55.2

### Operating system type + version
Windows 10

### Behavior
Slicer does not optimize top solid layer travels. 

#### Project File (.3MF) where problem occurs
[image]

[filterbucket.zip](https://github.com/supermerill/SuperSlicer/files/5974017/filterbucket.zip)
And there is how it prints. It use almost all time to bouncing around. 
https://youtu.be/_5kWiuDHL4M

burner-, 2021-02-16:
Any ideas howto get workaround for this. This printing fail again and again because of this bug.

supermerill, 2021-03-18:
I'm rewriting the gcode generation, it will take some time.