supermerill/SuperSlicer #959: When Ironing and Auto-Speed are enabled, all speeds are way too slow

NameOfTheDragon, 2021-02-14:
This is an issue I also reported in Prusa Slicer, see https://github.com/prusa3d/PrusaSlicer/issues/5082

### Version
2.2.53

### Operating system type + version
Windows 10 Pro x64

### 3D printer brand / version + firmware version (if known)
Creality Ender 3 with Klipper firmware

### Behavior

When ironing is enabled AND auto-speed is enabled, printing of the whole model is far too slow.

Steps to reproduce:
Print any model with the following settings enabled:
- Infill => Enable Ironing
- Infill => Ironing Type => All Top Surfaces

#### Project File (.3MF) where problem occurs

The problem occurs with this project file. [TPL Vacuum Seal Bag Nozzle.zip](https://github.com/prusa3d/PrusaSlicer/files/5490031/TPL.Vacuum.Seal.Bag.Nozzle.zip)

The problem can be seen in the generated gcode just after the beginning of layer 2:
[image]

As you can see, the feed rate is set to F159.118 (about 2.65 mm/s) which is crazy slow. It's even way slower than the default 15% setting for ironing speed (based on a default print speed of 100 mm/s).

When ironing is disabled, the following correct gcode is generated for the skirt:
[image]

NameOfTheDragon, 2021-02-19:
This is still an issue in version 2.3.55.2.

This was fixed in Prusa Slicer in one of the release candidates, 2.3.0-rc2 with comment: 

> I looked into it and the "Ironing is completely and utterly broken" is a bit exaggerated. In fact, it is autospeed that is broken by turning ironing on. Ironing produces very thin fill extrusions and that fools the autospeed, which then sets very small speeds. The ironing extrusions should probably be ignored for autospeed calculations. It is happening in autospeed_volumetric_limit function. It breaks not only the skirt, but all speed calculations. We will see how easy that'll be to fix.

It was fixed by this commit: https://github.com/prusa3d/PrusaSlicer/commit/30c8592b10954f09e78185fd06448cf7ce6064cd

supermerill, 2021-02-22:
merged

NameOfTheDragon, 2021-03-01:
@supermerill tested release 2.3.55.5 and confirmed working. Thank you!

closed 2021-03-18 by supermerill