supermerill/SuperSlicer #960: overextrusion in vasemode

hkr955, 2021-02-16:
### Version
2.3.55

### Operating system type + version
linux ubuntu

### 3D printer brand / version + firmware version (if known)
ender 3 + Merlin

### Behavior
* when using "spiral Vase" there is an heavy over
[Shape-Cylinder.txt](https://github.com/supermerill/SuperSlicer/files/5986987/Shape-Cylinder.txt)
extrusion next to the seam.
* It seems to be a Superslicer specific issue
[image]
And the print looks exactly like the flow shown at the scrrenshot.
* In Prusaslicer 2.3.0 it looks like this:
* [image]
And prints ok.
* _Steps needed to reproduce the problem_
** open application
** Add a shape
** go to print settings  and add the tick for "spiral vase"
** slice the cylinder
** select view "volumetric flow rate"

 

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`File -> Save` / `Save Project`)_

JohanAR, 2021-02-16:
Looks like #730

supermerill, 2021-03-18:
JohanAR is right, it's because of a little problem with G92 & absolute positioning.

I forgot to remove the G92 E0 from the custom gcode ->before layer. This shouldn't appear in the custom gcode, as it will mess up the internal extruder position.

supermerill, 2021-06-04:
reopen it if it's still an issue

closed 2021-06-04 by supermerill