supermerill/SuperSlicer #961: Worse Surface Quality than Before

derizk-cpu, 2021-02-16:
### Version
2.3.55.3 - 2.3.55.4

### Operating system type + version
Windows 10 

### 3D printer brand / version + firmware version (if known)
Prusa Mini (latest firmware)

### Behavior
* Finish quality of the walls is inconsistent, which results in worse quality in the actual print.

(I've been able to deduce that it's the result of changes from Dec 8, 2020. I believe it may result from dropping the minimum resolution, however changing the resolution in the slicer has no effect on the issue. Nothing in-slicer changed anything.)

* Create a cylinder in the slicer and look at the resulting g-code preview from an upward angle.
* Expected uniform-looking layers with only slight marks at the seam.
* Actually see non-circular extrusions which make the surface rough.

[image]
2.3.55.4 Close-up with 0.08mm layers and random seam placement

[image]
2.3.55.2 (stable version) Same settings

#### Project File (.3MF) where problem occurs
Applies to all models (create shape)

CCS86, 2021-02-23:
Check out this setting. I was playing around and found that zeroing it out reproduced your issue:

[image]

[image]

CCS86, 2021-02-24:
Playing around, values like 0.00005 worked just a touch better in trouble spots.

robthide37, 2021-02-27:
dos th nwst build fix this ? it is noted but does not say

derizk-cpu, 2021-02-27:
@robthide37 No, it's still an issue for me.

CCS86, 2021-02-27:
> @robthide37 No, it's still an issue for me.

Did you read my posts?

derizk-cpu, 2021-02-27:
@CCS86 I did. It's better, but it doesn't look the same as your result. I'm using the latest 2.3.55.5 version.

[image]
Resolution Set to 0, Rounding Precision at 0.00001, and Slice Gap Closing Radius at 0.049.
Prusa Mini 0.10mm Detail preset with seam position set to random.

CCS86, 2021-02-27:
You should still upload the project file. No one can recreate all your settings without it.

derizk-cpu, 2021-02-28:
@CCS86 

[Resolution Weirdness.zip](https://github.com/supermerill/SuperSlicer/files/6056122/Resolution.Weirdness.zip)

Open with the latest release and compare it with the last one.

CCS86, 2021-03-01:
It appears that having a randomized seam placement is causing this. Switch to aligned and it goes away. 

This makes sense, since the segment layout is relative to the seam, and if your seam is truly random, your segment start/ends will be random and cause this.

derizk-cpu, 2021-03-01:
> 
> 
> It appears that having a randomized seam placement is causing this. Switch to aligned and it goes away.
> 
> This makes sense, since the segment layout is relative to the seam, and if your seam is truly random, your segment start/ends will be random and cause this.

That _IS_ true, but what if I want to use random seam placement? In older versions the preview looked just like the aligned preview with the seam in random places (see first comparison). A circular path should make the same line regardless of where it starts and ends the seam. Furthermore, on more complex geometry, aligned seam position won't always have the seam in a vertical line, so you'll end up with similar artifacts to random.

Both PrusaSlicer and Cura handle random seams correctly.

CCS86, 2021-03-02:
I'm with you. I think handling segment alignment independently of seam placement would be best.

Just trying to add some context to help @supermerill

CCS86, 2021-03-10:
I found a situation where even with an aligned seam and tightened rounding precision, the surface is inconsistant:

[image]

[Tap Align - M10.3mf.txt](https://github.com/supermerill/SuperSlicer/files/6118875/Tap.Align.-.M10.3mf.txt)

robthide37, 2021-03-13:
try turning off reverse on odd ?

CCS86, 2021-03-15:
> try turning off reverse on odd ?

That's off for me.

robthide37, 2021-03-16:
oh

supermerill, 2021-03-18:
It's a very concerning problem. I tried to increase the precision, and it seems that it has the opposite result.

derizk-cpu, 2021-03-29:
@supermerill Unfortunately, putting 0 into "Resolution" and "Min segment length" doesn't really fix the issue. Even with the same settings, the older versions generate perimeters differently. The current versions don't align the perimeters correctly.

supermerill, 2021-04-24:
I think i found the bug
[image]
fix will come shortly

derizk-cpu, 2021-04-24:
@supermerill Thank you for fixing this! Out of curiosity, what was the issue?

supermerill, 2021-04-25:
I compared first.distance_to( first) instead of last.ditance_to(first)

CCS86, 2021-06-15:
I still seem to be seeing some of this effect on  2.3.56.5:

[image]

2.2.53:

[image]

It seems that 2.2.53 has much better small radii, but worse flat wall precision. But both have defects on both types of surface.

[Edging Coupler 90.3mf.txt](https://github.com/supermerill/SuperSlicer/files/6652398/Edging.Coupler.90.3mf.txt)

supermerill, 2021-06-15:
@CCS86 the "no perimter on bridge area" is the problem. Will be fixed in next release.

CCS86, 2021-06-15:
> @CCS86 the "no perimeter on bridge area" is the problem. Will be fixed in next release.
> also you have a very high "printer-> min extrusion size"

I used to run that setting at 0.005mm, but when you updated the behavior, I found that 0.25mm gave equivalent number of print segments to the prior 0.005mm setting. Perhaps the overly dense gap fill in the newer version skewed this comparison a bit. 

Setting down to 0.05mm still shows the issue though:

[image]

supermerill, 2021-06-15:
> Setting down to 0.05mm still shows the issue though:

I may repeat myself a bit but:

> the "no perimeter on bridge area" is the problem. Will be fixed in next release.

CCS86, 2021-06-15:
With "no perimeter on bridge areas" = remove perimeters:

[image]

With "no perimeter on bridge areas" = disabled:

[image]

It looks better, but still not fully consistent in stacking the print segments. Min extrusion length was set to 0.20.

It's also interesting to notice that this setting is somehow changing the extrusion width (look at the top gap fill)

CCS86, 2021-06-15:
With it disabled, you can still see some misalignment on a flat face:

[image]

(also present with min extrusion length = 0)

supermerill, 2021-06-15:
with 0.05 min extr dist

[image]

There is still some misalignment on a flat face, like you say. I'll see where it comes.

supermerill, 2021-06-15:
found the culprit: the curve_smooting is activated and is trying to smoothen some layers but not all of them... for whatever reason. Have to rewrite it.

CCS86, 2021-06-15:
Good catch! 

I hadn't used that feature in the past, but recently made a print where there was a very large radius curve. It came out with a noticeably faceted appearance. Once I tuned the curve smooth settings, it did a great job at shrinking those facets until they were not visible.

I never would have guessed that it was causing the flat face issue. 

Sorry to add to your backlog!

CCS86, 2021-06-26:
Hey @supermerill I just tested in 2.3.56.6 and it appears the "curve smoothing" disrupting flat faces bug is still present.

supermerill, 2021-06-26:
> Hey @supermerill I just tested in 2.3.56.6 and it appears the "curve smoothing" disrupting flat faces bug is still present.

extremely important: use a loose resolution setting, and hight model rounding precision, something like 0.02mm and 0.001). the whole point of the feature is to make nice curve on low-polygon model, so you must be sure it's as simple as possible.
Be sure your cutoff is good.

Also I can't know if a flat surface needs to be curved or not, so it will always be some mishap, that's inevitable.

CCS86, 2021-06-27:
Resolution=0.02
Rounding Precision=0.001

[image]

Resolution=0
Rounding Precision=0.00001

[image]

I think the second one looks slightly better.

But keep those settings and change 'curve smoothing precision' from 1.6mm to 0mm... perfect:

[image]

closed 2021-07-10 by supermerill