supermerill/SuperSlicer #964: When you save the r code, the name in Russian turns into gibberish, can you fix it?

AKlionRU, 2021-02-19:
### Version
2.3.55

### Operating system type + version
macOS Big Sur

When you save the r code, the name in Russian turns into gibberish, can you fix it? 
[image]

supermerill, 2021-02-22:
can't reproduce, i need your project file.

AKlionRU, 2021-02-22:
Here is a file with a Russian name 
[Коробка Natase.STL.zip] (https://github.com/supermerill/SuperSlicer/files/6023921/STL.zip)

supermerill, 2021-02-24:
can't reproduce on my windows
[image]

that may come from the "fold to acsii" function the file name passed by.
basically, it changes unicode character to ascii one to be sure the os can handle it as filename.

I don't know how to resolve it without creating more bugs.

AKlionRU, 2021-02-24:
The latest version of PrusaSlicer does not have this bug. there very simply solved this problem

supermerill, 2021-06-26:
fixed now. But the russian transaltion needs a native speaker to help me with translation.

closed 2021-06-26 by supermerill