supermerill/SuperSlicer #971: Feature request for expensive support - define interface extruder

bakaufman, 2021-02-21:
### Version
Doesn't matter

### Operating system type + version
Doesn't matter

### 3D printer brand / version + firmware version (if known)
Railcore dual extruder but applicable to multiextruder

### Behavior
I would like to define an extruder for my interface layer, which is $70/500g dissolvable filament. 
#### Project File (.3MF) where problem occurs
Not really needed.

cab404, 2021-02-21:
> ### Version
>
> Doesn't matter
> ### Operating system type + version
>
> Doesn't matter
> ### 3D printer brand / version + firmware version (if known)
>
> Railcore dual extruder but applicable to multiextruder [...]
[image]

cab404, 2021-02-21:
You can close this issue)

closed 2021-02-21 by bakaufman