supermerill/SuperSlicer #972: Cost & Weight is not displayed in Sliced Info Settings (and re-slice does not work)

DaStivi, 2021-02-22:
### Version
2.3.55.5 (updated right now to latest available 3b253f3bb4be85c000f5f5f878ba2e4cb396e246)

### Operating system type + version
Win10 21H2

### Behavior
added costs and weight of filament but does not show up in Sliced Info...
[image]

Also noticed that Re-Slicing doesn't work, so hitting either CTRL-R or even from the file-menu -> "(Re)Slice now" does not reslic it?! there has to be some change made, after the sliced ?!

### should look like:
[image]

supermerill, 2021-02-22:
[image]
could not reproduce,
 .3mf / project file?

neophyl, 2021-02-22:
You also need the density in the filament settings for the cost and weight to work. At least you do with Prusa Slicer now so I should think it is the same.

DaStivi, 2021-02-22:
> You also need the density in the filament settings for the cost and weight to work. At least you do with Prusa Slicer now so I should think it is the same.

that was exactly the issue... thx for pointing me too this... entering the density and costs and weight is displayed!

closed 2021-02-22 by DaStivi