supermerill/SuperSlicer #974: SuperSlicer build fail.

nanoplane, 2021-02-23:
### Version
2.3.55.5
### Operating system type + version
Mac OS Big Sur 11.2.1

### 3D printer brand / version + firmware version (if known)
N/A
### Behavior
Successfully ran cmake and make for deps.

Building SuperSlicer in xCode:
Can't complete BUILD_ALL due to multiple failures.

Build of SuperSlicer fails due to missing deps library: libwx_osx_cocoau_scintilla-3.1.a
Since deps build succeeded, I'm not sure where it's dropping the lib....

royal32, 2021-02-23:
I got past this by copying `deps/build/destdir/usr/local/lib/libwxscintilla-3.1.a` as `libwx_osx_cocoau_scintilla-3.1.a`. So the same file exists with both names. 
I'm not familiar enough with compilation to determine the root cause, though.

nanoplane, 2021-02-23:
So, did that and build succeeds but running it crashes on launch in  wxMacCoreGraphicsPenBrushDataBase::CalculateShadingValues
... bummer..

royal32, 2021-02-23:
Yup... tracking that issue [here](https://github.com/supermerill/SuperSlicer/issues/777), also mentioned [here.](https://github.com/supermerill/SuperSlicer/issues/936).

nanoplane, 2021-02-24:
So, looks like it's been solved. There's a comment that changes wxWidgets.cmake to use the wxWidgets 3.1.4-patched version for all Apple, not just arm..

royal32, 2021-02-24:
Yup, figured it out about an hour ago. I'd expect fixed builds to come pretty soon

ctilley83, 2021-02-24:
So it's building correctly, but the beta mac binaries posted a couple days ago simply crash on start? I'm experiencing the same thing on Big Sur 11.2.1

nanoplane, 2021-02-24:
Yep, crashes for me as well.